wled/WLED #492: Solid RGB Strip to dark

feindsender, 2019-12-20:
Hi,
i just tried out the Solid RGB Strip feature on my H801 Controller.
I think this part of the code should be outside of the switch (_type) statement.

```
    #ifdef WLED_USE_ANALOG_LEDS      
      //init PWM pins - PINs 5,12,13,15 are used with Magic Home LED Controller
      pinMode(RPIN, OUTPUT);
      pinMode(GPIN, OUTPUT);
      pinMode(BPIN, OUTPUT);
      switch (_type) {
        case NeoPixelType_Grb:                                                    break;
        #ifdef WLED_USE_5CH_LEDS
          case NeoPixelType_Grbw: pinMode(WPIN, OUTPUT); pinMode(W2PIN, OUTPUT);  break;
        #else
          case NeoPixelType_Grbw: pinMode(WPIN, OUTPUT);                          break;
        #endif
      }
      analogWriteRange(255);  //same range as one RGB channel
      analogWriteFreq(880);   //PWM frequency proven as good for LEDs
    #endif
```
Inside it will never get executed and the analogWriteRange will still be 0-1023.

Aircoookie, 2019-12-21:
Hi,
the code is not inside the switch block, but after it, it should be executed just fine...
Do you have proof that the analogWrite range is always 1024?

Happy holidays :)

feindsender, 2019-12-21:
Hi,
if i move the bracket from line 152 to line 134 the code gets executed fine.
Otherwise the code never gets executed because of the break in line 124 or 133.

Happy holidays 
Maik

Aircoookie, 2019-12-28:
You are totally correct! Was looking at the wrong switch block. Will fix ASAP!
Have a happy start into the new decade!

Aircoookie, 2020-01-04:
Fixed in 6e35b5ba8b6a39658836d761a3acfb688df6d1b0!

closed 2020-01-04 by Aircoookie