wled/WLED #493: button sync turns off after reboot

Etacovda, 2019-12-20:
Im using wemos d1 to trigger another via the button pins - finding after rebooting either that the button stops working. Works again when i re-enable on/off button in the sync interface menu. 

Actually what i am really trying to do is simply start and stop a present on a switch rather than a button for state awareness, and id rather do it with wires than wirelessly.

wchpikus, 2019-12-20:
Where is write last setup? Spiffs?
For me also this setup was not remember..

Aircoookie, 2020-01-04:
@Etacovda could it be that your second D1 is pulling the button pin to low? If it is not high on boot, the button is disabled. Let's also turn this issue into a feature request for switch support, which I could add soon!

@wchpikus The settings are saved in the EEPROM sector. If your ESP doesn't "remember" the config after reboot, probably the flash layout is off somehow. This can be caused by the flashing method. For best results, try the esptool command from the quick start [wiki](https://github.com/Aircoookie/WLED/wiki#quick-start-guide).

blazoncek, 2021-10-03:
Fully implemented as of 0.12.

closed 2021-10-03 by blazoncek