wled/WLED #494: Sync Settings showing Blank Screent

sogeniusio, 2019-12-20:
I noticed this a while ago but for some reason one of my Wemos boards the sync settings come up as blank. I would have to reboot the entire controller in order to get it up again. 

[image]

Aircoookie, 2019-12-21:
Hi, the blank page happens when the settings page could not be loaded for some reason. In the classic UI that is quite annoying since you have to reload the entire page to try again.
There are quite a few reasons why the page doesn't load, maybe the connection got interrupted or you ran out of memory.. Is this 0.8.5 or 0.8.6?

sogeniusio, 2019-12-21:
Looks like it's version 0.8.6

raoulteeuwen, 2019-12-21:
Could you try 0.9 (and, if that gives problems, 0.8.5)?

bobsiopener, 2020-02-25:
Hi,
I'm having the same issue with version 0.9.1 but it's only after I enable MQTT.
Once I enable mqtt, the device responds to the mqtt broadcast but i'm no longer able to access the sync settings, i have to factory reset the board if i need to change anything

Aircoookie, 2020-02-25:
@bobsiopener hi, that doesn't sound good at all. Are the sync settings not even accessible right after a reboot? What are your build settings?

bobsiopener, 2020-02-25:
@Aircoookie I'm using the build: WLED_0.9.1_ESP8266.bin
they are not accessible after a reboot, right after or otherwise. I just load a white page.
Let me know if there is any other info i can provide. thanks,
Bob

sogeniusio, 2020-02-25:
I wasn't able to test and check when the issue started originally for me
but I don't recall even enabling mqtt. And the issue still persists.

I'll try to run this newest version and see if it continues.

Aircoookie, 2020-02-25:
@bobsiopener thank you for the quick feedback! I believe the issue to be heap fragmentation, which is also related to the red disconnect warnings others have been getting in the main UI.
Just one last question: How many LEDs are you using? :)

bobsiopener, 2020-02-25:
@Aircoookie I have 5 different WLEDs one with 70led, 2 with 300led, and 2 with about 125led

bobsiopener, 2020-02-27:
@Aircoookie I don't know if you've committed a change to master for this fix, i compiled the latest and flashed it to an extra ESP8266 and I was able to access the sync page. Then I flashed one of the units that was having a problem over the air and I can access the sync page now. I had set my SSID and IP in the IDE before compiling but It's interesting that all the other settings i had set on the ESP are still there

Aircoookie, 2020-02-27:
@bobsiopener that's good news! Yes, I've made an effort to free 2kb of memory. This issue is quite clearly a manifestation of insufficient heap, so it is quite possible that the extra space made a difference!

Yes of course, the settings should always be retained between versions. Setting them in the IDE is also not officially supported, I'd recommend always settings them in the web UI :)

closed 2020-07-03 by stale[bot]