wled/WLED #495: Segments results in flickering

timdonovanuk, 2019-12-20:
Using an ESP32 + ws2811. Everything works fine, until I add a segment, and then those segmented LEDs flicker like crazy. They still get effects and are controllable, but with constant flicker in the area thats segmented.

Aircoookie, 2019-12-21:
Hi, you probably have overlapping segments. That is still not fully supported. Try shortening the Segment 0 (probably). Set its stop LED to the start LED of Segment 1, that will probably fix it :)

timdonovanuk, 2019-12-21:
Ah, I thought segment 0 represented all segments? As in, if you don't want to apply anything to segment 1 or 2, select segment 0 for all bulbs. I deleted segment 0 and its work now, thank you!

Happy holidays to you too btw :) Cheers

closed 2019-12-22 by Aircoookie

rawframe, 2020-01-01:
@Aircoookie Just to note, overlapping segments interfere with each other even when overlapping segments are deselected in the web interface.
With the current (non fully supported segments) release, to run an effect down the whole length on occasion I have to make sure there are no overlaps first an then apply the effect to each segment. This does not have have the desired effect because the effects are limited to their segment lengths meaning scan effect etc do not span correctly the entire strip length.
**It would easiest and more versatile to have unique segment setups also saved in each favourite preset. For example, fav1 has 2 segments (0-15 & 15-30) but fav2 has 3 segments (0-10 & 10-20 & 20-30.** and then loading these fav segment groups on the fly.

Aircoookie, 2020-01-06:
@rawframe deselecting a segment doesn't deactivate it, you'd need to remove it for that. The selected segments merely are the ones color or effect changes are applied to.

Yes, that's exactly what I'm planning. Presets that can save different segment configs are very important!