wled/WLED #496: Color platte does nothing

timdonovanuk, 2019-12-20:
I assume selecting a color palette is supposed to change the 3 set colours, but on 0.9.0-b1 (for me, anyway) selecting a palette does absolutely nothing.

Thanks!

Aircoookie, 2019-12-21:
Hi, yes palettes can be a bit confusing. There are a few effects (Solid is one of them) where the palette setting has no effect. Then there are others which only work with palettes and ignore the set color (like Palette, Fire 2012 or Colorwaves). That's is one of a few reasons I want to unify the concepts of colors and palettes in an upcoming release

Happy holidays :)

sr1sws, 2019-12-21:
Disclaimer: I'm a relative NOOB to WLED.

I think the palette settings only affect *some* effects. There's a list of effects that tells if they honor the palette setting.

The palette setting doesn't change the 1/2/3 colors under the color wheel. These are used for the effects that DON'T honor/use the palette setting.

Yeah, it's confusing. Best to play with it IMHO.

Good luck!

Steve

bitterdog, 2019-12-28:
For tri-color to set all three colors would be great. Like for 4th of July, a Red, White, and Blue motif would be perfect.

sr1sws, 2019-12-31:
I think you could do this with the CIRCUS effect, setting the secondary color to BLUE. CIRCUS sets the other 2 colors to RED and WHITE. 

Disclaimer: I have not tested this. 

Steve

tonyn0, 2020-01-01:
> I think you could do this with the CIRCUS effect, setting the secondary color to BLUE. CIRCUS sets the other 2 colors to RED and WHITE.
> 
> Disclaimer: I have not tested this.
> 
> Steve

Just tested and it works as you state.

closed 2020-05-10 by stale[bot]