wled/WLED #497: Holiday reduced communication notice 🎄🕯️

Aircoookie, 2019-12-21:
Hi everyone!

Just wanted to let you know that I'll be taking a few week's break over the holidays!

Knowing that I can't help myself, additions to WLED will probably still be made, but please be aware that I might miss your messages or take a long time to reply to them until about January 6th.

Have a merry Christmas and a great start into the new decade! 🎄🕯️🎆

Christian

bline79, 2019-12-21:
Thanks for all your hard work! Enjoy your time with family and friends :)

sansillusion, 2019-12-21:
Merry Christmas and Happy New year !

timdonovanuk, 2019-12-21:
My Christmas tree will look amazing this year and its all down to your hard work! Thank you so much :) <3

sr1sws, 2019-12-21:
Merry Christmas!

Mariu86, 2019-12-22:
Merry Christmas and Happy New Year Aircoookie! Happy holidays everyone!

dorinik, 2019-12-22:
Thanks for your work Aircoookie!
Thank you for making our winter holidays much more beautiful and colorful!
A much richer Christmas and New Year with your loved ones!
Happy holidays everyone!

StefanRu1, 2019-12-23:
Thank you so much!
Enjoy your holidays and have a Merry Christmas and a Happy New year !

sujitrp, 2019-12-25:
Merry Christmas and a Happy New year ! Enjoy

closed 2020-01-05 by Aircoookie