wled/WLED #498: LED's random/not lighting

Sebass85, 2019-12-21:
I could use some help. I dove headfirst into putting WS2812b led's along the roof line of my house. I have three nodemcu's, each controlling a 'zone', right, center, and left. I cannot get zones center and left to function. The right zone functions just fine. The first time I applied power to zones center and left, random led's along all the strips lit up solid various colors. The strips did not respond to any inputs from the nodemcu's. Now, no leds along the strips are lighting up. The strips are getting power, i checked with a multimeter at the end of the runs.

I have power running along each strip and injected at several points along each one, then to a 65amp power supply. The data wire goes from each strip then to the pin on its nodemcu. The nodemcu's are connected to the power supply to get power. I am at a loss because the right zone seems to work just fine, but I get pretty much nothing on the other two zones, and best I can tell each is hooked up the same way. Any help is appreciated, thanks.

sansillusion, 2019-12-21:
It may be the length of the data wire between your esp and leds. Do you use
level shifters?

sr1sws, 2019-12-21:
I ended up adding a 'sacrificial pixel' to my NodeMCU board to ensure proper functioning of the rest of the strip. At least the pixels I have (WS2811 12v) seem to be really picky about the signal. I originally tried just a level shifter w/out good/full success. I also injected power after 100 LEDs to get the last group to function correctly.

Hope this helps.
Steve

Sebass85, 2019-12-21:
I was under the impression that if the data wire was the problem I would at least get lit led’s that behaved weird, not get anything lit at all? I will try a sacrificial pixel though since it is a long run. When setting that up does that pixel require all three wires (ie power, grnd, and data) connected or will just splicing in a pixel with the data wire work? 
Thank you

sr1sws, 2019-12-21:
One other thought that stumbled into my head... on the misbehaving segments, are you feeding the data from the correct end of the strip?

sr1sws, 2019-12-21:
RE: Sacrificial Pixel - you have to provide power... the data chip needs it, the LED not so much. You can set WLED to ignore that first pixel if you want/need and it won't light, but it does need power.

Sebass85, 2019-12-21:
Ok thanks. I’ll try the sacrificial pixel and see what happens. I did make sure to properly orient the strips, the data is for sure going in the right end. Thanks I’ll Report back with the results.

Sebass85, 2019-12-21:
Alright I wired up two sacrificial pixels, spliced them into each power/ground line and still the same result. TheY don’t seem to have done anything. I put an image in of what the strips are now doing:

https://imgur.com/OMTvnMN

https://imgur.com/xDoaIyn

raoulteeuwen, 2019-12-21:
Afaik, your controller (NodeMCU) has to share ground with the power to your strip ... could that be the issue?

Sebass85, 2019-12-21:
https://imgur.com/CTOiSA6

This is how it is wired basically. When you say sharing the ground of the strip and the nodemcu is this what you mean or do you mean something different?

Sebass85, 2019-12-21:
Sorry forgot the sacrificial pixel. New image:
https://imgur.com/cVMxN6d

raoulteeuwen, 2019-12-21:
Mmm, looks pretty similar to for instance what is shown at https://youtu.be/tXvtxwK3jRk?t=1029

sr1sws, 2019-12-22:
Looks like it should work. Do the sacrificial pixels light? Is the blue led on the NodeMCU lit? It seems that weirdness is afoot! :(

sr1sws, 2019-12-22:
I feel your pain. Can you maybe swap around the NodeMCUs to confirm you have functional units? I know I killed one of mine very easy by shorting 12v into the data pin. :( Yeah, rookie mistake.

Sebass85, 2019-12-22:
The sacrificial pixels do not light. The nodemcus are all lit up.

Sebass85, 2019-12-22:
> Mmm, looks pretty similar to for instance what is shown at https://youtu.be/tXvtxwK3jRk?t=1029

Yeah his videos are what got me started

sr1sws, 2019-12-22:
Re: The sacrificial pixels do not light.... Then something is rotten. Bad flash of WLED, using wrong pin... something like that. I'm sorry, I'm out of ideas. If it was mine, I'd drag it back to my bench and poke at it until I figured out what the issue was.  Best of luck.
Steve

sr1sws, 2019-12-22:
> > Mmm, looks pretty similar to for instance what is shown at https://youtu.be/tXvtxwK3jRk?t=1029
> 
> Yeah his videos are what got me started

Me too.

Sebass85, 2019-12-25:
I figured it out. Several of the connectors I used closed, but did not pierce the wire insulation. So I had a few wires not doing anything. Thanks for trying to help!

sr1sws, 2019-12-25:
Glad you got it sorted! Merry Christmas!

Aircoookie, 2019-12-28:
Awesome you got it working just in time for Christmas!
Have a good start into the new decade!
(and thank you @raoulteeuwen, @sansillusion and @sr1sws for helping, it is seriously appreciated!)

closed 2019-12-28 by Aircoookie