wled/WLED #499: all my controllers reset to default

oneshot678, 2019-12-21:
all of my esp controllers running wled reset to the default effect after some time every night. and i have to re select a desired effect and color.

sr1sws, 2019-12-21:
I have not seen this. I have 0.8.6 running Christmas lights for a few weeks on NodeMCU. NodeMCU runs under constant power and I use the WLED timer and macro function to start and stop display.

I also have 0.9.0 b1 (or whatever it is) running 24x7 on a test without issues.   

Maybe check power and possible overheating?

Good luck and Merry Christmas!

vexter0944, 2019-12-21:
I'll second oneshot678's issue.  I'm running 0.9.0 b1 (or whatever it is) as well - I have it turn on via home assistant's integration and it will come on to my desired effect and at some point it will be working then reboot/crash and come back to solid amber/orange and just sit there until I do something with it.  

FYI - I'm running it on one of these - 
https://www.amazon.com/gp/product/B010N1SPRK/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

and have had to compile the code from scratch to get the led pin to be D5 because I'm using 12V and a level shifter.

sr1sws, 2019-12-21:
Well, I just put my set up into "production" so I'll see if I end up with trouble too.   I'm using these:
https://www.amazon.com/gp/product/B07HF44GBT/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

Did not compile, using the WLED_0.9.0-b1_ESP8266.bin file. Using pin 4 with a level shifter and sacrificial pixel to drive 147 WS8211 12v pixels.  Master power via 20A 12v supply from Amazon, power to NodeMCU dropped to 5v via buck converter.

DirkTas67, 2019-12-22:
> all of my esp controllers running wled reset to the default effect after some time every night. and i have to re select a desired effect and color.

I also have this problem, but not with all of my wled.

oneshot678, 2019-12-22:
does losing WIFI signal cause this? We have confirmed that others encounter this issue.

Aircoookie, 2020-01-04:
This definitely sounds like a crash/auto reset issue! @oneshot678 what ESP boards are you using with which binary? Are you using MQTT?

vexter0944, 2020-01-04:
No MQTT here - just as I commented above, using with HASS.  I did note that I did seem to notice some kind of memory leak via the 'sensor' in hass for the WLED free memory.  Like it would dip way down low - like the esp was running short on memory or something.  I was going to sit down and look and see if it coincided with the reboot/crash, but got too busy with the holidays and now they're all packed up.  I tried pulling history with HASS but looks like it's already gone :/

closed 2020-05-10 by stale[bot]