wled/WLED #500: Just discover 1 wled on app

tihoangyeudau, 2019-12-21:
I have 3 wemos d1mini and they all setup wled and connected to the same wifi network. When I use the Wled app on both Android or iOS phones, the app just discover 1 wled, and the other 2 cannot be found.

raoulteeuwen, 2019-12-21:
Can you access the Wemos's via a web browser, and make them connect to your router to get an IP address (so as to manage the WLEDs via a browser)?

tihoangyeudau, 2019-12-22:
> Can you access the Wemos's via a web browser, and make them connect to your router to get an IP address (so as to manage the WLEDs via a browser)?

yes, i can. i can see 3 ip of them and access to all of them via web controller

raoulteeuwen, 2019-12-22:
so just to make sure: everything works fine, except the app not detecting your WLEDs?

tihoangyeudau, 2019-12-22:
> so just to make sure: everything works fine, except the app not detecting your WLEDs?

Yes, sir

Aircoookie, 2019-12-22:
Thank you for reporting! Here are some things that might cause it:
- For how long did you try the in-app discovery? Some modules might need 15 seconds or so.
- Is the access point (WLED-AP) active? That might interfere with discovery.
- Is the mDNS field in wifi settings filled out and different on each controller?

Happy holidays :)

tihoangyeudau, 2020-05-04:
I solved this issue by the way that i reset wemod d1 mini , reset the app then try to discover again

closed 2020-05-04 by tihoangyeudau