wled/WLED #501: Wrong colors

sogeniusio, 2019-12-21:
For some reason I'm getting some discoloration or incorrect colors on my WLED.

Im running 0.8.9 and am using the WS2811. I have two strips in parallel coming from a Logic Level Shifter pushing the 3.3v data line from the NodeMcu To 5v.

I also have a capacitor parallel to my 12v input to stabilize voltage flow. But im seeing behaviors I'm not seeing with any of my other setups. Could this be a hardware issue? 

Here's a demonstration:
https://streamable.com/jnzbc

raoulteeuwen, 2019-12-21:
Could it be the RGB-setting needs adjustment? Look under Settings/LED setup/Color order

sogeniusio, 2019-12-21:
I thought that may be the case. I'm using WS2811 is the RGB config different from a WS2812?

sogeniusio, 2019-12-21:
I set it to BRG and we're good to go. Thanks.

closed 2019-12-21 by sogeniusio