wled/WLED #504: MQTT msg: getting some weird characters and causing colours not to be set properly.

adamdistefano, 2019-12-22:
I've been scratching my head for a few days as to why my colours were inconsistently set via MQTT and HEX via Homebridge + homebridge-mqtt-thing. I was able to narrow it down to WLED after compiling from source and setting the debug serial on. I did actually try tasmota and other firmware which worked fine.

I am seeing this in the logs when I send #ffffff via MQTT Explorer.

`MQTT msg: wled/wled5/col
#ffffff
MQTT msg: wled/wled5/col
#ffffffF⸮1ચ⸮⸮
MQTT msg: wled/wled5/col
#ffffffF⸮1⸮Ú⸮⸮
MQTT msg: wled/wled5/col
#ffffffCACAC⸮⸮⸮
Publish MQTT
MQTT msg: wled/wled5/col
#ffffffF⸮1⸮⸮⸮⸮⸮
MQTT msg: wled/wled5/col
#ffffff⸮⸮1⸮⸮⸮⸮
MQTT msg: wled/wled5/col
#ffffff⸮⸮1⸮'⸮⸮⸮
MQTT msg: wled/wled5/col
#ffffff⸮⸮1⸮@⸮⸮⸮
MQTT msg: wled/wled5/col
#ffffff⸮⸮1⸮Y⸮⸮⸮
MQTT msg: wled/wled5/col
#ffffffal
Publish MQTT
MQTT msg: wled/wled5/col
#ffffff
Publish MQTT
MQTT msg: wled/wled5/col
#ffffff
MQTT msg: wled/wled5/col
#ffffffP⸮Ą⸮{
`

It appears to be adding extra bits at the end and causing WLED to eventually set the colour wrong.

These bits are not being seen by any other client on this topic, it's only WLED. I've tried the latest release stable release and the latest beta (currently running 0.9.0-b1) without any success in resolving this.

Any ideas would be awesome!

adamdistefano, 2019-12-22:
So I managed to do a cheeky hack and "fix" my problem.

Just basically changed the mqtt.ino to strip those extra characters off the end before passing them to colorFromDecOrHexString() for processing.

I'll keep this open for discussion / response after the holiday period.

Aircoookie, 2020-01-04:
Hi!
Seems like the payload string is not correctly null-terminated. This probably happens when the sender does not append the null character to the payload. I will try to fix it!

closed 2020-05-10 by stale[bot]