wled/WLED #505: (Circular) LED matrix effects support

martijnvdwoude, 2019-12-22:
What would it take to develop support for circular LED rings effects? Is this something that WLED allows for?

This is the kind I'm talking about:
https://www.pcboard.ca/neopixel-ring

I'm specifically interested in doing effects like in this video:
https://youtu.be/nubyADJ7BFQ

sr1sws, 2019-12-23:
Disclaimer: WLED noob here.

I would think that you might do something like this with segments in 0.9.0_b1. 

Others may weigh in with different opinions. :) 

Steve

martijnvdwoude, 2019-12-23:
Just after opening this I installed 0.9.0-b1 and noticed the segments functionality! It seems like at least a big step in the direction to do something like this. I'll try to play around with it!

sr1sws, 2019-12-23:
If you set something up using segments, be aware that the full segment config can only be saved in preset #16. Just a heads up so hopefully you can avoid some confusion and frustration. ;)

Steve

Aircoookie, 2019-12-23:
I agree, segments are a good way to make this work. In the video they appear to have just connected the rings in parallel since they always run the same effect. The same could be done with WLED, just set the LED count to that of the largest ring.

yusufsulaiman, 2020-01-23:
hi. this is my led matrix ws2812b with pride 2015 effect. it is coool. thanks Aircoookie
https://www.youtube.com/watch?v=nGESMWByWhQ&t=39s

closed 2020-05-29 by stale[bot]