wled/WLED #506: Fairylight/Color Quick Selection

CaptainUns, 2019-12-23:
Maybe its just me, but I miss the quick selection color options that used to be located directly under the color wheel in the UI! Specifically, when it comes to approximating the incandescent/warm white/FastLED Fairylight color that used to be part of those 8 quick select colors. Its very difficult using the color wheel only to dial this color in just right. Is there a workaround for this or the possibility that those 8 quick-select colors might be brought back in a future build? Or better yet, maybe a way to input a FastLED color name (such as Fairylight) and have the current color (whether primary, secondary or tertiary) change to that color? Especially with effects like Twinklefox, on 0.8.6 I had it set up with both primary and secondary colors using that Fairylight color to approximate classic twinkling Christmas lights and it looked great, but using just the color wheel I can never get the two colors exactly right. Anyone have any suggestions or thoughts on this?

Aircoookie, 2019-12-23:
I agree, I should bring these back!
What I tried was to streamline the new UI as much as possible, but the quick color selectors are something that was quite helpful.
Another thing I am considering is a simple input field that let's you set a hexadecimal color code (like #FFF0A0)

Happy holidays :)

sr1sws, 2019-12-25:
I cast another vote for the color "quick picks" to return! But I do like 0.9.0_b1. ;)

CaptainUns, 2020-01-02:
Maybe a combination? It would be great to be able to enter specific hex color codes. Would there be a way to add customization the quick selection buttons? Maybe in a similar way to saving presets? Find the color you want either through the color wheel or by entering a hex color code, then save that color as one of your quick selections? There are a couple colors (i.e. Fairylight, sports team colors) that I use frequently and being able to save and quickly access those colors when programming effects would be fantastic!

sr1sws, 2020-01-02:
I also like the idea of HEX or decimal for the RGB values - but if I really want to set them precisely at this time, I use the HTTP API interface.

I've also started "collecting" HTTP API strings with specific settings for future use. E.g. I just created one for Mardi Gras with purple and teal.

raoulteeuwen, 2020-01-02:
> I've also started "collecting" HTTP API strings with specific settings for future use. E.g. I just created one for Mardi Gras with purple and teal. I also have one for UF Gator colors - although their published color spec looks 'bleh' on LEDs... so I tweaked it.
> 
Is it an idea you share those someone for others to enjoy?

sr1sws, 2020-01-02:
Sure... just a string that I fire at WLED from the browser:

You can find the HTTP request codes linked in Github. These may or may not make sense for your installation. I keep these in OneNote for easy use. 

One observation: I couldn't get a save into Preset 16 (&PS=16) to work from HTTP. I know 16 is unique in being able to store segments. I just switched to preset 1, which worked. Other presets probably work too.

**Christmas** - cycle thru presets 1 - 16
http://192.168.254.200/win&T=1&P1=1&P2=16&PL=1&CY=1&PT=30000

**Autumn** - presets 1 - 4
http://192.168.254.200/win&T=1&P1=1&P2=4&PL=1&CY=1&PT=30000

**UF**
http://192.168.254.200/win&T=1&R=255&G=160&B=0&R2=0&G2=0&B2=255&FX=37&SX=127&IX=127&FP=0

**Mardi Gras**
http://192.168.254.200/win&T=1&R=128&G=0&B=128&R2=0&G2=128&B2=129&FX=18&SX=115&IX=124&FP=0

**White Sparkle**
http://192.168.254.200/win&T=1&R=255&G=255&B=255&R2=168&G2=168&B2=168&FX=20&SX=228&IX=133&FP=0&PS=1

sr1sws, 2020-01-02:
HTTP request API reference found here: https://github.com/Aircoookie/WLED/wiki/HTTP-request-API

tonyn0, 2020-01-04:
Another +1 for quick color selectors!

commit 4f7dc5be (2020-06-22): Added Sunrise nightlight mode and more UI features

-   Added Sunrise nightlight mode
-   Added Chunchun effect
-   Added `LO` (live override) command to HTTP API
-   Added `mode` to `nl` object of JSON state API, deprecating `fade`
-   Added light color scheme support to web UI (click sun next to brightness slider)
-   Added option to hide labels in web UI (click flame icon next to intensity slider)
-   Added hex color input (click palette icon next to palette select) (resolves #506)
-   Added support for RGB sliders (need to set in localstorage)
-   Added support for custom background color  [... cut]
closed 2020-06-22 by Aircoookie