wled/WLED #508: Request: Fade On / Fade Off time

StefanRu1, 2019-12-23:
Hi,

i have a request.
I use Transition Time: 1000 ms
But what would be great would be two additional settings for Fade On and Fade Off.
So that i can set the timing when the light is switched on and off.

Behind this request is the usecase that i control my Wled controlled stripes with a motion sensor and MQTT.
What i want to achieve is that the light fades quick in when motion sensor triggers.
After this trigger the light stays on for a certain amount of time as long as no new motion is detected.
Everytime a new motion trigger is send from the motion sensor the timer starts again.

If the timer is over the light should fade out very slow, so that i have time to move again to activate the motion sensor trigger.

I use On and Off triggered by motion sensor and timer. An option to  set the Fade On and Off time would be great.

Thanks and best regards,
Stefan

Aircoookie, 2019-12-28:
Hi Stefan!

It should be possible to get that to work with the current API by using the `TT` (transition time) command :)

You can try:
`A=255&TT=1000` for on, and
`A=0&TT=60000` for off (it will fade down over the course of a minute, set shorter if you'd like)

I hope it will work well!

Have a great start into the new decade!
Christian

StefanRu1, 2020-01-03:
Hi Christian,
hope you had a good start in the decade.

That sounds perfect. 
I will try it in my installation via MQTT.

Thanks for your help and all the effort.
Stefan

closed 2020-01-04 by Aircoookie

iz3man, 2020-02-12:
Could you please tell me exactly how you do this? I want to do something similar. It's a night light. When motion is detected it shall fade in quite fast (maybe 500ms) and then fade out once motion detection is over. 
What is the MQTT command to send in that case? Thanks a lot.