wled/WLED #511: Turn segment on or off with api

SenexCrenshaw, 2019-12-24:
I just want to be able to turn a segment on or off and retain its color values. I cant seem to find this. From what i have seen is on and off is global. I could set the color to all 0s but then i wont know the color once it is turned back on. Am i missing something or is this not possible?

Aircoookie, 2019-12-28:
Hi!
Currently the easiest way to do that is to set the `stop` of the segment to 0, which will disable it.
The only issue with it is that your application needs to know what the original stop value is to turn the segment back on. I might consider adding on/off and brightness/opacity to segments at some point :)

jnewland, 2020-02-10:
Hi! I'd love to be able to turn a specific segment on or off without changing its configuration and would be willing to spend some time contributing this functionality. Any suggestions before I dive in? 😁

Aircoookie, 2020-02-12:
@jnewland Hi! The Segment object in FX.h has the currently unused `opacity` property.
My vision would be to blend whatever is already on the LED strip with the segment's effect if opacity is less than 255.
Basically, you could have two overlapping segments, one with opacity 255 and the next with 128. This would make it possible to properly overlay those segments, but there is still a lot of work to be done for overlapping segments to work.
The problem is that for that to function, all effects need to be able to run at the same framerate, and they also shouldn't rely on the current LED state (`getPixelColor()`), because they may have other segments overwriting them and all LEDs would be cleared to black between each frame.
See [this list](https://github.com/Aircoookie/WLED/projects/4) :)
This opacity can then also be used as an individual segment on/off and brightness control.
I believe we could implement a rudimentary version right now where opacity 0 means the segment is off (it isn't even being rendered at all) and any other value would be just on for now.
What do you think?

jnewland, 2020-02-12:
Hey @Aircoookie thanks for following up! I read a bit of code yesterday morning and am glad to hear that the unused `opacity` property is slated for this use: I agree that it could be used a a rudimentary on/off control in a first pass of this work. I also agree that _blending_ opacity with the provided pixel values makes as a target over the long term.

Unfortunately, the requirements of the project I'm working on have changed such that this feature isn't a requirement anymore, so I'm not sure I'm going to be able to dedicate time to it in the near term. I'm truly sorry about that!

Aircoookie, 2020-02-12:
@jnewland thanks for getting back to me! No worries, I hope I'll be able to add the feature soon.
Best of luck with your project :)

DutchmanNL, 2020-03-18:
would be happy to see a possibility to just turn on/off specific segments + 1 for this feature

ikrowni, 2020-03-26:
add me to the list i'd love to see this feature in a future update.

Aircoookie, 2020-05-02:
Added, will be part of release 0.10.0 tomorrow!

closed 2020-05-02 by Aircoookie

DutchmanNL, 2020-05-03:
Nice thank you will update my adapter 😊