wled/WLED #512: STOP WORKING-Many and many errors and not working-0.9.0b1 on nodeMCU

pepeEL, 2019-12-24:
Hi
First i write WLED on my nodeMCU and after some days stop working but can not connect by IP.
Again flash nodeMCU by other new device. This also works some days and not GUI by IP open but function like POWE ON and other not working. Configuration by GUI works ok. Bu i can not power on and control my leds. Looks what error show... IT is two of more errors,
[image]

[image]

pepeEL, 2019-12-24:
[image]

Aircoookie, 2019-12-28:
Hi, sorry for the issues!
They should already be fixed in source, also see https://github.com/Aircoookie/WLED/issues/453
Releasing a new binary in a few days :)

closed 2020-10-12 by Aircoookie