wled/WLED #513: HTTP Port# for webhook req

chitrangad, 2019-12-24:
Hi Aircoookie,

I'm trying to setup webhook via IFTTT to control the lights over internet.
I understand I have to enable port forwarding to accomplish this. However, I don't know which port# WLED uses for json post. Any info on this would be appreciated.

Thanks again for the wonderful app.
Merry X'Mas!

Aircoookie, 2019-12-28:
Hi!

Thank you, I hope you have/had a merry Christmas as well :)
Like any HTTP connection, WLED uses port 80 for the JSON API. Most routers will allow you to do a port forwarding to any external port you'd like (for example, you could set external port 2000 to map to WLED port 80). In that case, you'd use port 2000 in IFTTT.

Hope it helps!

chitrangad, 2019-12-29:
I feel dumb that never tried the default port!
Thanks again!

closed 2019-12-29 by chitrangad