wled/WLED #514: Preset issue v9-b1

Biothefirst, 2019-12-24:
First I want to say how much I love wled! Thanks for making this code. 

So I'm new to wled. And I have it all set up and working. 
Running a string of 300 sk6812. 
But when I go to make presets with effects and pallets. It does not record correct for 1-15. 
But on 16 it does. 

My understanding of 16 was that it would save segments too. 
And 1-15 was 1 segemnt plus effects and color pallets.

Is my understanding of this correct?

Thanks for any guidance.

raoulteeuwen, 2019-12-24:
Slot 1-15 currently allow you to store configured effects while your strip is configured as '1 segment'. If you would like to use the new segments option and configure some segments and effects, those currently can only be stored in slot 16.

Biothefirst, 2019-12-24:
And that is how I understood it to work. But it does not. 

I only have one segment. But when it records it just records a solid color at an intensity. 

But if I record to 16 it records correctly

raoulteeuwen, 2019-12-24:
Can you share, step by step, how you try to set a certain effect and save it to a slot?

Biothefirst, 2019-12-24:
Sure.

1 open phone app 
2 select color palette and intensity
3 swipe left and select effect 
4 set speed and size of effect. 
5 swipe left twice to favorites
6 check box to turn on saving mode. 
7 select a number between 1-15
8 unselct save mode 
9 push number of saved preset 

Effect not show just solid color with intensity. 

If I change step 7 to be preset 16 it works and saves the effect. 

No segemnts. Just one long strip

raoulteeuwen, 2019-12-24:
Thank you. What does the segment tab say? Mine shows "Segment 0" without a check mark in front, followed (next line) by Add segment.

Biothefirst, 2019-12-24:
Ah ha! I must have made a segment at one time and then deleted it. 

My segement was listed as number one. I had deleted segment 0.

Biothefirst, 2019-12-24:
Thanks for all the help raoulteeuwen!

closed 2019-12-24 by Biothefirst