wled/WLED #517: WLED-AP never appears

paularmstrong, 2019-12-25:
* esp32s
* binaries tried: 0.8.5, 0.8.6, 0.9.0-b1

No matter what I do, the wireless access point never shows up. The logs from esphomeflasher are completely blank, as if the device never actually starts up.

Logs via esptool.py
```
$ esptool.py write_flash 0x10000 ~/Downloads/WLED_0.8.5_ESP32.bin
esptool.py v2.8
Found 4 serial ports
Serial port /dev/cu.usbserial-A900HH38
Connecting........_____....._____....._____..
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 3c:71:bf:f9:fb:48
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1027744 bytes to 593610...
Wrote 1027744 bytes (593610 compressed) at 0x00010000 in 52.7 seconds (effective 156.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

Logs via esphomeflasher
```
Using '/dev/cu.usbserial-A900HH38' as serial port.
Connecting........_
Detecting chip type... ESP32
Connecting...

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 3C:71:BF:F9:FB:48
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dio
 - Flash Frequency: 80MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 6.6s
Compressed 15856 bytes to 10276...
Writing at 0x00001000... (100 %)Wrote 15856 bytes (10276 compressed) at 0x00001000 in 0.2 seconds (effective 531.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 143...
Writing at 0x00008000... (100 %)Wrote 3072 bytes (143 compressed) at 0x00008000 in 0.0 seconds (effective 1330.0 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 3957.8 kbit/s)...
Hash of data verified.
Compressed 1034832 bytes to 591312...
Writing at 0x000a0000... (100 %)Wrote 1034832 bytes (591312 compressed) at 0x00010000 in 14.3 seconds (effective 578.9 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
```

NOTE: The line from the wiki is very vague for someone unfamiliar with arduino
> ESP32 (you will need to have a bootloader installed, you can flash the Arduino blink example to do that)
I believe I flashed the device with the "blink" example (though the blink example apparently is not ready out of the box and took some googling to figure out you need to add something like `#define LED_BUILTIN 13` to the file before installing). I did this, it installed, then I went back and tried to install WLED again, same results.

Is there something wrong with what I'm doing? Is a step missing from the instructions?

raoulteeuwen, 2019-12-25:
Afaik, Aircookie recommends esptool .... did you see this part on https://github.com/Aircoookie/WLED/wiki:

    If you experience issues, run this command before trying write_flash again (Note: this will erase all settings stored on the ESP!)

esptool.py erase_flash

    You can alternatively use my basic HTTP OTA updater sketch and upload the binary! This requires the Arduino IDE and ESP8266 core installed.

Did you try both?

paularmstrong, 2019-12-25:
I tried all of that. Eventually, after I gave up, it connected while I was trying a manual connection...

One major note: OS X displayed the setup page as a dialog. It did not require navigating with a browser.

Once I got it connected to my wifi, it seemed to work. However, it seems there may be some other issue, as the wifi doesn't seem to _stay on_. It will work for 5-10 minutes, then disappear from the network. I had esphome flashed on this device previously and it was working just fine, so I suspect something in WLED is causing issues.

paularmstrong, 2019-12-29:
Went back to esphome and the wifi works flawlessly. What can I do to help get WLED working?

Aircoookie, 2020-01-04:
Hi and sorry for the issues!
You could try reflashing the bootloader once again before using esptool (for example by uploading Arduino Blink sketch).
I will try to find what else might be causing your issue :)

paularmstrong, 2020-01-04:
> You could try reflashing the bootloader once again before using esptool (for example by uploading Arduino Blink sketch).

There seems to be a code error in the Blink sketch. Could you recommend something different that could be uploaded that will actually compile?

Mariu86, 2020-01-06:
If You have a error maybe it's something wrong with your settings.

raoulteeuwen, 2020-01-06:
@Aircoookie any chance you contacting the esphome dev to exchange how you both handle wifi would be beneficial?

paularmstrong, 2020-01-06:
> If You have a error maybe it's something wrong with your settings.

Never used Arduino app for anything else. Haven't touched any settings. What could I have missed configuring?

raoulteeuwen, 2020-01-06:
> Never used Arduino app for anything else. Haven't touched any settings. What could I have missed configuring?

Did you find and use the Compile settings at https://github.com/Aircoookie/WLED/wiki ?

paularmstrong, 2020-01-06:
> > Never used Arduino app for anything else. Haven't touched any settings. What could I have missed configuring?
> 
> Did you find and use the Compile settings at https://github.com/Aircoookie/WLED/wiki ?

Noob to ESPs here... what do I do with those? I used the "Quick Start Guide" higher on the page and followed the instructions under "I just want to use WLED! (install release binary)"

raoulteeuwen, 2020-01-06:
> Noob to ESPs here... what do I do with those? I used the "Quick Start Guide" higher on the page and followed the instructions under "I just want to use WLED! (install release binary)"

Because you mentioned "Never used Arduino app", and a remark was made about settings, i thought you might need those compile settings. But those, afaik, are only needed when you use Arduino IDE to flash stuff to your NodeMCU. You can find the Arduino IDE at https://www.arduino.cc/en/main/software . You could try it, it'll definitely make you learn something ;-) . Installation instructions for the Arduino IDE desktop app can be found at https://www.arduino.cc/en/Guide/HomePage . After which you could use http://help.blynk.cc/en/articles/512105-how-to-install-blynk-library-for-arduino-ide to flash the Blynk sketch to the NodeMCU.

paularmstrong, 2020-01-06:
Blynk? @Aircoookie and docs keep saying "Blink", which is built-in to the Arduino IDE and what I'm referencing when I say there's a compile issue.

raoulteeuwen, 2020-01-06:
You're right, sorry, i'm mistaken...

Aircoookie, 2020-01-06:
Hi, sorry for the trouble getting it to work. I realized that even flashing the blink sketch requires you to install the ESP32 arduino core, which is not hard, but also not trivial.
I've extracted the Arduino bootloader to a .bin file for you: https://drive.google.com/file/d/1NQrBwb5WJz6WSEbotW-6NrmiLJ6Hc04F/view?usp=sharing

Just flash it to your ESP32 using
```
esptool.py write_flash 0x0 .\esp32_bootloader.bin
``` 
and everything should work! (if it doesn't, try running the regular ESP32 flash command from quickstart again)

Aircoookie, 2020-02-27:
@paularmstrong is this resolved?

paularmstrong, 2020-02-27:
I switched to a D1 mini using 0.9.x and it's working

closed 2020-02-27 by paularmstrong

oxihormesis, 2020-06-09:
I just had this problem with an ESP32S- erased, flashed, flashed bootloader in different combinations and finally saw Aircookie's response. Thanks! I flashed the esp32 bootloader binary AFTER flashing wled binary and it worked!

me-dex, 2024-10-20:
Update: I got WLED to compile and flash with PIO without issue for a ESP32-S3-DevKitC-1-N8R2. It seems that the pre-built binary produced for ESP32S3 by WLED is missing several components being flashed alongside the binary that PIO handles automagically

```
Writing at 0x00000000... (100 %)
Wrote 14768 bytes (10148 compressed) at 0x00000000 in 0.3 seconds (effective 410.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 127...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (127 compressed) at 0x00008000 in 0.1 seconds (effective 488.1 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 640.5 kbit/s)...
Hash of data verified.
Compressed 1367072 bytes to 920468...
Writing at 0x00010000... (1 %)
Writing at 0x0001793f... (3 %)
Writing at 0x0001b867... (5 %)
Writing at 0x0001f8dc... (7 %)
Writing at 0x000238e2... (8 %)
Writing at 0x000278dc... (10 %)
Writing at 0x0002b8d7... (12 %)
Writing at 0x0002f9cb... (14 %)
Writing at 0x0003ab4d... (15 %)
Writing at 0x00044bed... (17 %)
Writing at 0x00050422... (19 %)
Writing at 0x0005faf0... (21 %)
Writing at 0x000655ad... (22 %)
Writing at 0x0006b2fb... (24 %)
Writing at 0x00070ee7... (26 %)
Writing at 0x00076b18... (28 %)
Writing at 0x0007bf0f... (29 %)
Writing at 0x00081adb... (31 %)
Writing at 0x00087bf2... (33 %)
Writing at 0x0008d507... (35 %)
Writing at 0x00092db9... (36 %)
Writing at 0x00099484... (38 %)
Writing at 0x0009eab7... (40 %)
Writing at 0x000a4cc3... (42 %)
Writing at 0x000aab21... (43 %)
Writing at 0x000b0602... (45 %)
Writing at 0x000b6b90... (47 %)
Writing at 0x000bc876... (49 %)
Writing at 0x000c2a33... (50 %)
Writing at 0x000c803a... (52 %)
Writing at 0x000cdb09... (54 %)
Writing at 0x000d2b67... (56 %)
Writing at 0x000d7bfc... (57 %)
Writing at 0x000dcb4b... (59 %)
Writing at 0x000e1cd0... (61 %)
Writing at 0x000e6c2c... (63 %)
Writing at 0x000eba9b... (64 %)
Writing at 0x000f0c3e... (66 %)
Writing at 0x000f58dc... (68 %)
Writing at 0x000fa927... (70 %)
Writing at 0x001001b2... (71 %)
Writing at 0x00105997... (73 %)
Writing at 0x0010a754... (75 %)
Writing at 0x0010f878... (77 %)
Writing at 0x00114841... (78 %)
Writing at 0x00119cee... (80 %)
Writing at 0x0011f0d3... (82 %)
Writing at 0x00124b07... (84 %)
Writing at 0x0012a283... (85 %)
Writing at 0x0012fbf3... (87 %)
Writing at 0x00135051... (89 %)
Writing at 0x0013d252... (91 %)
Writing at 0x00145c2a... (92 %)
Writing at 0x0014be0a... (94 %)
Writing at 0x00151379... (96 %)
Writing at 0x00156bbd... (98 %)
Writing at 0x0015c918... (100 %)
Wrote 1367072 bytes (920468 compressed) at 0x00010000 in 8.0 seconds (effective 1368.2 kbit/s)...
Hash of data verified.
```

me-dex, 2024-10-20:
[image]

me-dex, 2024-10-20:
It appears that the data being flashed at 0xe000 is the factory OTA image (boot_app0.bin) which is what gets booted first on a factory reset. Looks like PIO might be flashing it with a stub. It would explain why WLED wasn't booting properly previously. Would probably help if a WLED binary was built with these components built in like with Tasmota or if it was more documented

DedeHai, 2024-10-20:
Manual flashing requires you to manually flash all partitions. The web installer should take care of that for you, if the board configuration is supported, as does VSC PIO.

me-dex, 2024-10-20:
> The web installer should take care of that for you, if the board configuration is supported, as does VSC PIO.

I can confirm that the web flasher is still broken for the ESP32-S3-DevKitC-1-N8R2 board. Results in the behavior described in https://github.com/Aircoookie/WLED/issues/3677

me-dex, 2024-10-20:
It looks like the `esp32s3dev_16MB_PSRAM_opi` is a standard build for WLED, not one of the more unusual hardware configs described in the web-installer issue - https://github.com/Aircoookie/WLED-WebInstaller/issues/22

https://github.com/Aircoookie/WLED/blob/1ff667b7eff07ebad100e48eb871c698eab11a71/platformio.ini#L45

Seems like more of a general issue with ESP32S3 builds/the web installer even for a standard board config

blazoncek, 2024-10-20:
> I can confirm that the web flasher is still broken for the ESP32-S3-DevKitC-1-N8R2 board. Results in the behavior described in [#1212 (comment)](https://github.com/Aircoookie/WLED/issues/1212#issuecomment-729142252)

S3 was not supported at the time of that issue. Unrelated.
S3 requires appropriate binary and bootloader depending on the type of flash and PSRAM. There are at least 6 different combinations of that.

If you think web installer is broken, please open an issue there. WLED in itself is fully working on any type of S3 we have encountered so far. Main reason for non-working installation is incorrect flashing procedure or incorrect binary type for chosen chip's flash/PSRAM combination.

me-dex, 2024-10-20:
> S3 was not supported at the time of that issue. Unrelated.

So that's still the behavior I'm getting from the latest beta builds with a web flash/manual flash with the `esp32s3dev_16MB_PSRAM_opi` build

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cd97a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2920
entry 0x403c98d8
```

The stable build of 0.14.4 flashed from the web flasher seem to be resulting in a different behavior (resolved by flashing over PIO):

```
SP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
...
```

It does seem related to https://github.com/Aircoookie/WLED-WebInstaller/issues/22 but extended to include standard board configs as well

softhack007, 2024-10-20:
> It does seem related to [Aircoookie/WLED-WebInstaller#22](https://github.com/Aircoookie/WLED-WebInstaller/issues/22) but extended to include standard board configs as well

The details you find in 
https://github.com/Aircoookie/WLED-WebInstaller/issues/22#issuecomment-2080430946 are  actually true for all -S3 configurations. I'm not sure what is a "standard S3 config" for you - actually wethese main configurations exists 
* flash size: 4mb, 8mb, 16mb (32mb not supported yet in WLED)
* PSRAM: none, less than 8mb (qspj mode), 8mb or more (opi mode).
So its 3x3 = *9* major configurations with different bootloader, partitions.bin, and firmware image. Only the "app0 partition is always the same. Unfortunately we cannot provide you with 2 or 3 firmware that "run on any S3".

And ... depending on who you ask ... depending on the month of the year ... you'd probably get different opinions on what is "standard"

I would strongly suggest that this discussion is put into the place where it belongs: https://github.com/Aircoookie/WLED-WebInstaller/issues

softhack007, 2024-10-20:
A good overview of known -S3 configurations is here:

https://espeasy.readthedocs.io/en/latest/ESPEasy/ESPchips.html#quad-octal-spi-mode

softhack007, 2024-10-20:
Btw, the discussion here went _very much_ off-topic already. The original problem report was about some esp32S (classic esp32) not running with "dio" firmware.....

RDJunc, 2024-12-01:
Looks like I am to be added to the long list of those that have this problem.

Background:
I am not new to the ESP32 and its many flavors but I am new to WLED.  (AMAZING!!!)  So far, I have just used it with controllers that I bought.  Two years ago, before I new about WLED, I designed a Christmas tree star with smart LEDs.  I wrote firmware that included several effects but no where near what WLED can do.  My primitive version accepted Bluetooth commands from a phone to change the effect.  This year, I designed a new version that uses an ESP32-MINI.  (I did this to avoid the issues that may be associated with the S and C versions.)  I was careful about handling the strapping pins properly.  The control pin is GPIO4.

Once built, I installed my original firmware and all was well.  I use Visual Studio with the Visual Micro add-in as my IDE.  It all uploads and runs just fine.

I have tried pretty much all of the ways to flash WLED onto the ESP32.  This includes the WLED installer, esptool, and the Espressif flash tool.  No joy.  They all seem to flash but WLED doesn't show up in my network list.

Here is the response from esptool...

`C:\Users\RJuncosa\Desktop\esptool-v4.2.1-win64>esptool.exe -p COM6 write_flash 0x10000 WLED_0.14.4_ESP32.bin
esptool.py v4.2.1
Serial port COM6
Connecting...........
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-U4WDH (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 1c:69:20:13:f7:d0
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00010000 to 0x00149fff...
Compressed 1284976 bytes to 852357...
Wrote 1284976 bytes (852357 compressed) at 0x00010000 in 78.5 seconds (effective 131.0 kbit/s)...
Hash of data verified.`

One things that I did notice is that when I flash my code with Visial Studio (which uses esptool) the reset after the load is clean.  The transmit and receive LEDs on the board are off and the star runs its default effect.  However, when I use esptool from the command line, the reset is not clean.  The receive LED is lit.  When I open a terminal, this is what I get...

`rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401468b8: 21a10041 291f812c 00106403
Core 1 register dump:
PC      : 0x401468be  PS      : 0x00060d30  A0      : 0x1d75dce8  A1      : 0x3ffb1d70  
A2      : 0x00000010  A3      : 0x00000002  A4      : 0x00060120  A5      : 0x00000001  
A6      : 0x00060120  A7      : 0x00000000  A8      : 0x80146854  A9      : 0x3ffb1d50  
A10     : 0x00000000  A11     : 0x00060123  A12     : 0x00060120  A13     : 0x3ffb1efc  
A14     : 0x000000ff  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x401468bb:0x3ffb1d70 0x1d75dce5:0x3ffb1d90 |<-CORRUPTED
`
Of course, any assistance would be greatly appreciated.

blazoncek, 2024-12-01:
Please use WLED [forum](https://wled.discourse.group) or [Discord](https://discord.gg/FMvz8nzmat) for help and support questions.