wled/WLED #518: Support for TM1814 RGBW leds

ryanjjennings, 2019-12-26:
I have some TM1814 leds I am looking for a way to control.  Any chance it will be added to WLED?

This was about the only info I could find on it.

Thanks for the help,
Ryan

Aircoookie, 2019-12-28:
Hi Ryan!
The easiest way to get them supported would be to ask the Neopixelbus creator Makuna if he can support them. He made the excellent library WLED uses for driving the LEDs!
I'd do it, but I just opened an issue with him to support WS2801 already.

Have a good start into the new decade!

Aircoookie, 2020-01-04:
Seems like this could be happening soon!
https://github.com/Makuna/NeoPixelBus/issues/325

ryanjjennings, 2020-04-19:
Just checking in to see if the Neopixel changes made it into WLED?

Aircoookie, 2020-04-19:
They hadn't yet, I just added the define to be able to easily select TM1814 in WLED!

commit 6a5b757a (2020-04-19): TM1814 support (closes #518 )
closed 2020-04-19 by Aircoookie