wled/WLED #519: Add option to show pass WiFi

pepeEL, 2019-12-26:
Hi
Good option will be add to setup WiFi checkbox to show password wifi. This is good option and sometimes need.

pepeEL, 2019-12-28:
Is any chance to add this feature request ?

Aircoookie, 2019-12-28:
Hi,
while I agree that it would be convenient sometimes, it is quite the security risk to transmit the WiFi password in plain text every time the settings are opened.

I will think about it, but I don't think that it is justifiable unless HTTPS and a user login (similar to what you suggested in #520) are added.

blazoncek, 2022-02-21:
Closing due to insecure nature of exposing passwords over HTTP.
No excuse should be good enough for exposing passwords IMO, even over HTTPS connections.

closed 2022-02-21 by blazoncek