wled/WLED #520: Add secure by password

pepeEL, 2019-12-26:
Good option will be add option to secure access to configuration by password. This will be very need.

pepeEL, 2019-12-28:
Is any chance to add this Feature Request ?

Aircoookie, 2019-12-28:
Hi!
For wifi and security settings, this is already supported using the "OTA lock" password in security settings. This will block updates and optionally access to the wifi settings. I might consider adding password support to the entire UI or the entire settings.
This has already been suggested here, so I'm closing this issue: https://github.com/Aircoookie/WLED/issues/308

closed 2019-12-28 by Aircoookie

pepsonEL, 2020-11-10:
Hi
Is any news about add option to security panel UI ?

GeneArnold, 2021-01-03:
I just wanted to add my support for password protection. I have two daughters that will be playing around with each others night lights if I can't password protect each one. I really would be very helpful. Thank you very much!