wled/WLED #521: Wifi-settings get lost after power lost

tuffelh, 2019-12-27:
Hi,
i have issues with WLED in combination with a Wemos D1 mini.
Everything works except that all settings including the Wifi Settings get lost after the Wemos lost power.
Any known issue I can prevent?

Aircoookie, 2019-12-28:
Hi,
how did you install WLED?
Likely the flash layout is off somehow so settings aren't preserved.
I'd recommend going with the `WLED_0.9.0-b1_ESP8266.bin` binary and the esptool command from here: https://github.com/Aircoookie/WLED/wiki#quick-start-guide

I hope we will be able to solve the issue :)

tuffelh, 2019-12-29:
Hi Aircookie,
thanks for your reply. It was indeed the flashtool.
Now everything is working as expected! Thanks a lot for creating this project and your support!

I think I will need to order some more wemos :-D

closed 2019-12-29 by tuffelh