wled/WLED #522: Nodemcu v3 WS281b connection issue

1kohm, 2019-12-27:
Hi!

I'm struggling witn basic setup of WLED ( version 0.9.0-b1) running on Nodemcu v3 and LED strp of 300 led WS2812b. The issue is quite strange:
- When I connect ws2812 led ring to pin D4 (https://www.adafruit.com/product/1586) it's working like a charm
- when i connect 300led strip to pin D4 ws2812b then WLED won't even boot (can't ping device)

I use same PSU (5V, 12A) for both LED's, I've tried to change nodemcu boards, tried also to change PSU to 5V 10A. 

I've read carefully issue #197 and tried all provided guidelines like using LogicShifter (3,3V -> 5V) but the result is the same - WLED don't want to start. 

Looking forward your answers,

raoulteeuwen, 2019-12-27:
Afaik, one RGB-LED needs (when on max bright white) 0.06A. 300 LEDs need 18A. I would look into whether your power supply can deliver what is needed for your strip... Check the power at the end of the strip.

1kohm, 2019-12-27:
Hi,

Thanks for answer. I thought that as well, therefore I cut my LED strip to 1m (60led) to bypass PSU issue. Still after connecting LED strip data pin to D4 on Nodemcu - esp8266 stop responding. Reset button don't work. When D4 pin is disconnected then ESP8266 is back responding. 
Best,

raoulteeuwen, 2019-12-27:
The LED-strip, nodeMCU and power share the same ground?

1kohm, 2019-12-28:
Yes, they have the same ground. 
My current connection looks like that: 
[image]

When Nodemcu is on without D4 pinn connected I can ping WLED but when I connect D4 data pin I'm loosing connection, also after restart. 
I added 1000uF capacitor to 5V line in parerall to provide more smooth power but it did not helped.

gwaland, 2019-12-28:
Are you using the right pin on the level shifter?  this diagram shows it's setup to receive data from the LED (RXI to RXO).  You might try it so that the nodemcu can send data to the LEDs (TXI-TXO).  At least in the old sparkfun level shifters the TX line is bidirectional and the RX only flows high to low.   I'm not sure if that's the issue but that might help.

1kohm, 2019-12-28:
This level shifter is not exact what i have. My has LV1 - HV1 ... LV4 - HV4 so is isn't bidirectional? 
Therefore do I need to use bidirectional level shifter? 
I've also flashed nodemcu v3 with different image "WLED_0.9.0-b1_ESP8266_ledpin3.bin" that use "RX" pin for LED strip an the result is the same. NeoPixel ring works like a charm but when connecting LED strip - WLED crash. 
Now I'm testing LED strip with 5 LEDs only and different power supply (5V, 3A) to eleminate power supplu issue. 
Maybe my nodemcu v3 with esp8266mod is not compatible somehow?

Mariu86, 2019-12-29:
I use the level shifter in 2 of my setups and I have only 1 power supply for the led and NodeMCU, I connected the + in the vin pin and - to the ground next to vin and for the 3.3 v for the level shifter I have taken the opposing pins from the NodeMCU 3.3v and ground, for the led strip and level shifter I have another wire from the power supply. Everything its working ok, my level shifter has only hv( high voltage 5v)  and LV ( low voltage 3.3v) write on it .

Def3nder, 2019-12-29:
You could try to RIP out the level shifters and connect the data wired directly to pin D4.
If the wire to the first Pixel is short and the data wire not too small then this does the job perfectly.

The 1000 uF capacitor is a very good idea anyway - leave this in to smooth Vcc.

1kohm, 2019-12-31:
After days of testing I've found the issue! :) 
It was not WLED code related.
I used second LED strip connection for 5V and Data pin and it worked! That was the last thing that I've tried after reflashing 10 ESp8266, changing connections, etc...
Can someone explain me why original connection was not working properly ?

Thanks for your all feedback.

Mariu86, 2019-12-31:
Maybe if you make some pictures to see exactly your setup we could help you. Happy new year !

1kohm, 2020-01-01:
Happy new year! 
Now my setup is now straightforward and no level shifter for pin D4 for data.
Now I'll start testing.
[image]

Aircoookie, 2020-01-04:
Feel free to reopen if you need more help!

closed 2020-01-04 by Aircoookie

bachoo786, 2021-02-04:
hey @Aircoookie  I have an issue with my wled. I have 300 leds in a 5m strip so I am using a 5V 18amp power supply. I have injected power every 1.5m to make the leds bright so in total I have injected power 3 times.

Everything works fine but sometimes I see the first few leds where I am feeding the power from light up. Is that normal ? I was thinking of adding a 1000uf capacitor but I am not sure if I need to add the capacitor every few meters where I am injecting power into the led strip?