wled/WLED #523: When using MQTT the board goes into boot loop

DennisH1978, 2019-12-27:
Hi there.
I got WLED on my board. in the Web UI everything is fine, easy to install and to use, thx very much for this cool program. But even if i want to run this by MQTT in the IoBroker, the board gets into boot loop. If i turn the adapter in the IoBroker off, it runs stable. The password is correct, both ports are the same. There is no overload or anything else. The IP is static and single used, no other device has the same IP.
This issue happens on wemos and nodemcu. It happens on the last three versions WLED.
It happens on different ports. It happens in different times, sometimes after a few minutes, sometimes directly after turning the adapter on. I don't know whats wrong here. For further Information i show you the Log of IoBroker, where you can see, that the reconnect happens always after 5-6 Seconds.
[image]

Aircoookie, 2019-12-28:
Hi!
Sorry to hear that there are issues with MQTT... Could you send me a screenshot of the WLED MQTT settings section? One more thing you could try for testing is leaving the group topic field empty, that way there will be less load. Hopefully we can get this worked out!

blub001, 2020-01-02:
Hi,

I have the same issue! 
I am using sonoff basic R3 with WLED. 
When enabling MQTT, evrything seems to be fine, till you want to turn on and off with iobroker. 
When turning WLED on via MQTT and again off, it toggles on/off like:
2 minutes off
10 seconds on
5 minutes off
10 seconds on
1 minute off
10 seconds on

WLED toggels on/off for 10-15 minutes till it stays continious off. 
When turning WLED on it is stable on. -> no toggeling.

I left group topic empty. -> no changes

Aircoookie, 2020-01-04:
Are you using the default MQTT port of 1883 or another one?

blub001, 2020-01-04:
I am using 1886. 
by the way, nice project! I have two devices running with WLED. But only one with MQTT due to that issue. 

I have tested to send to both devices the same MQTT command (off) and after a while one board toggeld the other was stabel. They do not toggle at the same time.
I also tried to use a third sonoff basic R3 board but I had the same issues.
A good friend of mine had the same issue.

closed 2020-05-10 by stale[bot]

T0rax, 2020-06-15:
Hi, no solution for this issue? I just started with WLED (0.10) Build 2005030 for an ESP-01S and ioBroker. When enabling the MQTT on ioBroker and WLED the first time, all is fine. When rebooting the WLED controller, i hang in a bootloop. I can see one or two successfull ping respons until ESP reboots. My Wifi device also shows me reconnecting the esp again and again.
When i stop the mqtt instance in ioBroker, the controller will stop rebooting and works as expected.
When i restart the mqtt, it will reboot again. If i stop the mqtt, delete the device in the mqtt and start again, it will rebuild the mqtt device again. Now the esp works fine again until next reboot.

I also tried the WLED adapter for iobrker, but this is not "realtime". I want to use the mqtt push information from the WLED esp to sync other RGB devices, which are not WLED.

Aircoookie, 2020-06-15:
@T0rax Sorry for the issue! Please try disabling the Alexa emulation in Sync settings, that often helps with bootloop issues.