wled/WLED #525: Set color brightness through http request?

avraham, 2019-12-28:
Thanks for this software. I love it

I followed the wiki steps and tried sending the request myipaddress/win&A=255, which seems to work for the 'master brightness' but not for the color brightness.

I include a picture to explain myself better

[image]
Thanks

Mariu86, 2019-12-28:
For the color you need to send her colors like A=255&R=60&G=127&B=80.

avraham, 2019-12-28:
Thanks for the help.
I'm actually sending an hex value and it picks the right color from the wheel color picker, but the bar i pointed out with the arrow it always reset to half.

Is there a parameter for that?

These are my parameters: CL=H00538C&A=255

Aircoookie, 2019-12-28:
Hi!
The bar sends the brightness of the RGB color! That means you modify it by sending different RGB values as @Mariu86 pointed out. `CL=H00538C` is about half brightness, so that is correct. You can try `CL=H0097FF` for example, that is the same color, but using full RGB brightness.

avraham, 2019-12-29:
Got it. :)
Thank you so much

closed 2019-12-29 by avraham