wled/WLED #526: ESP32 Wroom 32U - how flash by this BIN

pepeEL, 2019-12-28:
Hi
I buy this model PCB ESP32 Wroom 32U
https://www.aliexpress.com/item/4000103411061.html?spm=a2g0s.9042311.0.0.1ec64c4dXRvukd

How i can flash it by your firmware from realse BIN for ESP32 ?

Aircoookie, 2019-12-28:
Hi!
Have you tried this already? https://github.com/Aircoookie/WLED/wiki#quick-start-guide
Basically you need to use esptool and make sure to write to 0x10000. Also depending on the board, you might need to flash a simple arduino sketch like Blink to it before that.

pepeEL, 2019-12-28:
Ok to close. i use ESPHome-Flasher-1.2.0-windows-x64 but i must put in flash mode my ESP32 but press RST+BOOT, then release RST and after 1 secund relase BOOT.

closed 2019-12-28 by pepeEL