wled/WLED #529: 2 Pins connection pinout

duartefranco, 2019-12-28:
What are the pinout on nodemcu to connect the ws2081 2 pins (clock CI and data DI)?

Aircoookie, 2019-12-28:
Hi!
For APA102 2 pin LEDs, the default pins are 0 for clock and 2 for data.
WS2801 are unfortunately not supported right now, see https://github.com/Aircoookie/WLED/issues/178
Hopefully I will make some progress on that soon :)

feindsender, 2019-12-29:
Hi,
I added support for the WS2801 to the Neopixelbus Library and NpbWrapper.h.
I will attach the changend files in this post.
The NpbWrapper.h Part is a bit ugly because the WS2801 only supports RGB and not RGBW.
Maybe you find a better solution. It is working pretty nice with my LED Strips and also the Strings.
I also had to change 

```
void WS2812FX::setPixelColor(uint16_t i, byte r, byte g, byte b, byte w)
{
  i = i * (_disableNLeds+1);
  if (_locked[i] && !_modeUsesLock) return;
  if (IS_REVERSE) i = SEGMENT.stop -1 -i + SEGMENT.start; //reverse just individual segment
  byte tmpg = g;
  switch (colorOrder) //0 = Grb, default
  {
    case 0: break;                  //0 = Grb, default
    case 1: g = r; r = tmpg; break; //1 = Rgb, common for WS2811
    case 2: g = b; b = tmpg; break; //2 = Brg
    case 3: g = r; r = b; b = tmpg; //3 = Rbg
//    case 3: g = b; b = r; r = tmpg; //3 = Rbg
  }
```
in FX_fcn.cpp for my Strips witch are RBG.
The Strings are RGB and working fine without the change.

[ws2801.zip](https://github.com/Aircoookie/WLED/files/4007903/ws2801.zip)

Def3nder, 2019-12-29:
The case structure seem to be okay this way.
In the current code the option '3' will result in "BGR" and not - as written in the comments - in "RGB".
So maybe we need to extend the values by one to get GRB, RGB, BRG, BGR and **RBG**.

feindsender, 2019-12-29:
Or extend by two values. This should cover every possible combination.
RGB
RBG
GRB
GBR
BRG
BGR

GaryOkie, 2020-01-07:
came here trying to find an answer to why "GBR" color sequence isn't supported in the WLED UI.     (The post above this was the only reference to that sequence I could find).

It is an Alitove WS2811 (SM16703) LED strip that requires this GBR sequence.

Aircoookie, 2020-01-07:
#561 was also just opened. I'm just going to add all the color combinations at this point :)

Aircoookie, 2020-01-14:
Added all the color orders to master!

closed 2020-01-14 by Aircoookie