wled/WLED #530: Sync affecting other controllers with different UDP port

Gordus-Arrechus, 2019-12-29:
Hi, my lights been working OK so far but notice an issue today.

I have 2 nodemcu with wled 0.9.0-b1 one is the master that send effects, brightness, etc to the slave. The master has 300 leds and the slave 100.

The slave is configured to receive UDP data and all is working ok.

BUT, today i was working with a 3rd node mcu for other project and it started to sync with the master of the first 2. i changed the UDP por and same thing. 

Is there something i need to change so the first two are always in sync but do not affect the other?

I will likely to have several controllers around the house and all will have different effects and settings therefore i need to solve this first.

Appreciate your help.

Cheers

Aircoookie, 2019-12-29:
Hi,

does the issue persist after you reboot the board where you changed the port?
The change will only take effect when restarted :)

If you do not want to restart, unchecking all the sync checkmarks should take immediate effect.

Thank you for using WLED and have a great start into the new decade!

Gordus-Arrechus, 2019-12-30:
Hi, yes the issue persist.

The only way I have now to avoid this is to disable sync on all my other controllers and leave the 2 (already in production) syncing between them.

Thanks again for your help.

Cheers

Aircoookie, 2020-01-04:
Interesting, thank you for the feedback! Will have to look into that.

closed 2020-05-10 by stale[bot]