wled/WLED #531: wifi_set_phy_mode(WIFI_PHY_MODE_N);

dzikus, 2019-12-29:
Please add “wifi_set_phy_mode(WIFI_PHY_MODE_N);”. This will give possibility to use WLED with new routers/ap.

Aircoookie, 2020-01-04:
Could consider this! Have you tried this or is there other evidence that it might solve issues?
Phy mode B could also be interesting as an option as it may increase range (G is in-between and the default)

dzikus, 2020-01-06:
I’m using openwrt with mwlwifi drivers. After connecting WLED without wifi_set_phy_mode(WIFI_PHY_MODE_N); my 2.4 is completely broken. There is many issues on github related to this https://forum.openwrt.org/t/mwlwifi-and-esp-devices-update-wrt32x/34156/5 for example. I’m using many esp devices in my home (esphome, espeasy, shelly), only WLED is not using N.

pety3bi, 2020-08-04:
Possibly it would be great to see this configurable from webUI

closed 2022-10-21 by Aircoookie