wled/WLED #532: Bin file link

wchpikus, 2019-12-29:
Is is possible to add last compiled bin (default d4 pin) to apk or www?
It will be easier to flash new version.
Now i have to searching it..

Def3nder, 2019-12-29:
Hi @wchpikus ,
you can find the .bin files in the releases section.
This is the default distribution channel.

https://github.com/Aircoookie/WLED/releases

wchpikus, 2019-12-29:
Is it include all changes?
I watching this project about 3 weaks..or i have to compile it myself?

Def3nder, 2019-12-30:
The master is very close to the 0.9b2, so which changes are you waiting to be included in the .bin ?

Aircoookie, 2020-01-04:
@Def3nder is correct, you can always download the latest binaries on that page. Yes the version is from 3 weeks ago and there are a few minor changes, so I will release a new version shortly!

closed 2020-01-04 by Aircoookie