wled/WLED #533: Fine for days, now crashes/unresponsive after 20 mins

timdonovanuk, 2019-12-29:
Was working fine, now after 20 mins of working the lights go out or reset to default and the board is uncontactable. Have reduced the brightness to see if its power brown outs and tried just static single colour but same thing. Have to cycle power to get the board back. Odd that it worked fine for a few days.

Def3nder, 2019-12-30:
Hi @timdonovanuk ,
this sounds like a “bad ESP board”. Did you try to exchange the ESP ?

sr1sws, 2019-12-31:
Thermal shutdown?

Aircoookie, 2020-01-04:
Hi!
That indeed sounds like it could be a hardware issue, considering the board had been working well for a few days. What ESP are you using and how did you install WLED? Is MQTT enabled?
You could also try if completely erasing flash and reflashing helps.
Thank you :)

closed 2020-05-10 by stale[bot]