wled/WLED #534: Analogue LEDS - White Channel Not Working in E1.31 Mode

AndrewApi, 2019-12-30:
Hi
Just tested new code for "Fix analog write range" release "WLED 0.9.0-b2 (build 1912291)", works great now full brightness on LED Strip.

I am using RGBWW analogue strip, WLED controls all 4 channels when RGBW is selected, but in E1.31 mode I can only control the 3 RGB channels, I can not get the white channel (CH4) to light at all. Is this a possibility to correct please.

Thank you for all your hard work on this great project.

Def3nder, 2019-12-30:
Hi @AndrewApi ,

did you try to set the option „ Auto-calculate white channel from RGB: “ in the LED Settings ?
Does this change anything ?

AndrewApi, 2019-12-30:
Hi Def3nder

Thanks for the reply.
This takes away the White channel fader in GUI, it does not change anything when trying to control the White channel via E1.31.

Aircoookie, 2020-01-04:
Hi!
Unfortunately, any channels but RGB are not supported in E1.31 mode currently.
The reason for that is that most E1.31 sending software assumes 3 channels (RGB) per LED.
I might consider adding an "E1.31 layout" or similar option, that would allow you to control the white channel via E1.31.

pille, 2020-02-13:
i'd like to implement E1.31 RGBW modes analog to those introduced with #665.
does it make sense to wait until you figured out emulation in #573?

Def3nder, 2020-02-13:
Hi @pille, is there any software that can send RGBW via E1.31 ?

pille, 2020-02-13:
it should be possible to just add another channel for the white value.
i'm using QLC+ with my own fixture definition.

Aircoookie, 2022-10-21:
DDP now supports white channel, but there is basically no software that supports it (yet)

closed 2022-10-21 by Aircoookie