wled/WLED #535: User Interface stops responding.

perozonp, 2019-12-31:
This is happening in both 0.8.5 and 0.8.6
It works fine for several hours and then stops responding. Is like the web server is halted because still responds to mqtt commands (Home Assistant can still control it). All I have to do to "recover" it is cycle the power, and again works for several hours more (less than 24). While in that "UI halted" state the "power" button still works.

Aircoookie, 2020-01-04:
Hi!
Sorry for the issues! Did you already try the 0.9 beta? (but i do not expect it to improve anything if the issue happens in both 0.8.5 and 0.8.6).
Seems like the webserver might not be restarted correctly after a WiFi disconnect. I will need to investigate that. What ESP are you using and how good is the signal?

perozonp, 2020-01-04:
Just installed 0.9 (much nicer user interface!!!). Will test for a few days and let you know. Wifi signal is strong but I know that connection fails maybe once or twice a day (I think one of my devices is causing trouble with the wifi) so I agree with your initial diagnosis (web server not restarted correctly after a wifi disconnect). I'm using a nodemcu 8266 with 4megs and the binaries I have used are the "plain" 8266 version.

rmerrel, 2020-01-06:
Hope it is not bad form to comment on this issue but I'm experiencing the same issue (or at least a very similar version of it) and I hope that my symptoms may provide additional clues.  
I ran version 0.8.5 for several weeks with no issues.  I used the UI via Chrome on my windows based computer.  After running for several weeks with no real problems, one day it simply stopped connecting to the UI.  Eventually I flashed a different nodemcu with the 0.9 beta and can access the UI on that one.  By the way, I like the new interface.
0.9 has been working fine for several days.  However, today, I cannot access the UI via my browser.  I'm setup with a static IP and while troubleshooting I confirmed that my IP is still the same and connecting via wireless.  I have several simple bookmarks on my browser set to use the HTTP API (like http://192.168.0.45/win&PL=2).  These still work, they switch to my different pre-sets.
I noticed in the FAQ  that you suggested doing a reset for a different problem using <IP>/reset.  I did this and I do get a response in the browser after the reset.  It says:
Rebooting now...
Please wait ~10 seconds...
But after waiting for quite some time, I still cannot get the regular UI to respond.  It simply takes me to a white/empty screen in my browser.
The problem that I am seeing is somewhat different than what perozonp is seeing in that if I power the nodemcu off and back on, I still cannot get to the UI.  I have wondered if a hard power off/on is contributing to the problem.  My system is setup with a light switch that controls power going to the nodemcu and LEDs.  I simply turn the power on and off.  Could that be contributing to the problem?  Could it be leaving the WLED software on the nodemcu in a bad state?
I have verified the same problem in both Chrome and IE from two different computers.

rmerrel, 2020-01-13:
After experimenting with this for a while, I think my problem may be that the nodemcu is simply too far from my modem.  This results in a poor wireless connection that may work at times and may not work at other times.  I have ordered a wifi range extender and will test the system in it's original configuration once it arrives.

perozonp, 2020-01-13:
After about a week with 0.9 I can confirm the issue is still there (and is not poor WIFI range). Since in my case it seems to be the WEB server that goes down for unknown reasons, It would be very helpful to have some MQTT command to either restart the WEB server or restart the whole unit. My concern with this issue is I plan to install several of them in places difficult to access, so I'll need an option to reset/restart (currently is only available through the web server). But, of course, if this issue can be solved then much better.

rmerrel, 2020-01-16:
Sorry to have muddied the original issue reported by perozonp.  My issue was/is not the same as his.  I have confirmed that the problem I had was simply a poor wifi signal to the nodemcu.  I added a wifi extender to my network and it seems to be working perfectly.

closed 2020-05-22 by stale[bot]