wled/WLED #536: Signal not working

noblesse1234, 2019-12-31:
Hi guys,

I followed the video on youtube on how to flash and get WLED working.

At first the LED strip was working as intended, changing lights upon clicks on the app.

However, when i tried to set it up properly the next day, it no longer works.

Tried a new NodeMCU & new led strip but still the same.

The connection to the app and the custom IP is working, it's connected when the NodeMCU is powered on.

But changing effects, on/off & etc on the app does not change anything to the led strip (it's lighted up but nothing changes).

The strip that I am using is a 24V WS2813 (as claimed by the seller) led strip. Attached is a link to the said led strip:

[https://www.mediafire.com/view/kuq3rv7zwjtmj25/led_strip.png/file](url)

I connected 24V to the strip, and then from the strip's additional 2 wires, i used a DC-to-DC converter to convert 24V to 12V and feed it to NodeMCU baseboard HW-389. The signal wire i connected it straight to NodeMCU's D4.

I even tried to power up the NodeMCU using just microusb without other power source and connected the signal to D4 but it's still not working.

I tested with several new NodeMCU and led strip to no avail.

But i swear it was working fine the first time i tested.

Thanks.

raoulteeuwen, 2019-12-31:
You're feeding the NodeMCU 12V? Not an expert, but thought those could only run via 3.3 or 5V.
But if your NodeMCU is somehow working, i would check whether your powersupply (how much A?) can feed your strip (how much LEDs?). And whether the data-connection is o.k. (i assume your nodeMCU and strip share the same ground as they are both powered from the same power supply).

noblesse1234, 2019-12-31:
> You're feeding the NodeMCU 12V? Not an expert, but thought those could only run via 3.3 or 5V.
> But if your NodeMCU is somehow working, i would check whether your powersupply (how much A?) can feed [...]

The baseplate for the NodeMCU accepts up to 24vdc. It converts to the appropriate voltage for the NodeMCU.

Power supply is sufficient, as the led strip is lighting up, but just not changing effect when i changed it in the WLED app.

I used jumper wire for the data connection, changed several new jumper wires but still the same.

Yes they are sharing the same ground when powered via DC-to-DC converter.

wchpikus, 2019-12-31:
Show the diagram or picture of connection

noblesse1234, 2019-12-31:
> Show the diagram or picture of connection

Here you go:

https://www.mediafire.com/view/i6loci7c27g5yvj/Led_Strip_ESP8266.png

I encountered a strange problem with this setup, the WLED app can't detect new light when the NodeMCU is powered this way.

But when i power NodeMCU via PC MicroUSB instead, the WLED app detects it. I tried connecting the LED Strip's data line to D4 while the NodeMCU is being powered by MicroUSB, but the LED Strip is still not responding to changes made in the WLED app.

wchpikus, 2019-12-31:
First..change power the led strip..i do not thing that is power by 24V..could you send link about this strip? Where you saw the 24V power?

Second, connect esp using usb and just connect gnd and d4 for strip.
If esp working when you using usb, you have problem with this 24-12V dc-dc..btw..your edp havd to lost 7V on ldo..it is lot of power.

noblesse1234, 2019-12-31:
Hi, it's 24V. Check this picture i posted earlier:

https://www.mediafire.com/view/kuq3rv7zwjtmj25/led_strip.png

It's written 24V in the LED Strip.

Not working still. I have checked the output of the 24VDC to 12VDC with digital multimeter and detected nothing wrong with it. Check:

https://www.mediafire.com/view/cut0br7esua85zk/12v.png

Is WLED simply incompatible with 24V? But it's weird because it worked the first few times i tested it.

wchpikus, 2019-12-31:
Esp has no connection to power of strip..
Show link where you bought this strip or datasheet.
Maybe levels of data is 5V or higher.

noblesse1234, 2019-12-31:
> Esp has no connection to power of strip.. Show link where you bought this strip or datasheet. Maybe levels of data is 5V or higher. W 31 grudnia 2019 12:15:40 noblesse1234 <notifications@github.com> napisał:
> […](#)

Unfortunately, the link or the seller does not have the datasheet. It's a chinese site: https://item.taobao.com/item.htm?spm=a1z09.2.0.0.627d2e8d9UL2St&id=555584673186&_u=vu9ubnuda70

Does that mean a level shifter is needed?

Def3nder, 2019-12-31:
The strip needs to be connected via D4 **+ GND** to the ESP.
Could you add a jumper wire from ESP GND to GND of the strip ?

You may have too many DC-DC converters in this setup (24 to 12 and 12 to 3.3).

noblesse1234, 2019-12-31:
> The strip needs to be connected via D4 **+ GND** to the ESP.
> Could you add a jumper wire from ESP GND to GND of the strip ?
> 
> You may have too many DC-DC converters in this setup (24 to 12 and 12 to 3.3).

Sorry, reconfirming just in case.

Do you mean connecting ESP via microusb for power, and then connecting ESP's D4 + GND to led strip and then powering LED strip through 24V PSU?

This may sound stupid but do i need to use DC converter for the GND?

wchpikus, 2019-12-31:
No ...connect esp to usb, gnd from esp to gnd to strip, d4 to data in strip.
Power esp from usb not from psu..test it

wchpikus, 2019-12-31:
Btw...you showed pic of strip..do you have any information, where do you bougt it?
If this strip using ws leds, this leds needed 5V of power, so, this strip should have some psu on board..or this is not ws led.

wchpikus, 2019-12-31:
https://www.google.com/url?q=http://www.normandled.com/upload/201605/WS2813%2520LED%2520Datasheet.pdf&sa=U&ved=2ahUKEwi2gpfLguDmAhUsxosKHdgPCUkQFjAAegQICBAB&usg=AOvVaw2zUNY7Ssf_3Vt85c_SpaSK

wchpikus, 2019-12-31:
Write that the chip is soldered on strip.

sr1sws, 2019-12-31:
Personally, I'd confirm that the 24v to 12v converter ground is shared on both sides... check w/multimeter to confirm zero ohms.

NodeMCU output level is 3.3v. LED strips I've used want 5v for signal. I think it just depends on if 3.3v  will work. I use a level shifter.

I've had various issues driving strips without using a "sacrificial pixel" to provide the drive signal - mostly if the signal is traveling more than a few inches from NodeMCU to the strip. 

Good luck!
Steve

sr1sws, 2019-12-31:
Re: powering NodeMCU w/24v - I agree it's marked that 24v is OK, but I also read instances where they have died when powered that high. Personally, I use a cheap buck converter and feed it 5v. The on-board SMD regulator will work much less hard dropping 5v to 3.3v than 24v or even 12v. Just my opinion.

noblesse1234, 2019-12-31:
> No ...connect esp to usb, gnd from esp to gnd to strip, d4 to data in strip.
> Power esp from usb not from psu..test it

Did that, same result.

I asked my seller again, he said it uses 1903 IC.

> Re: powering NodeMCU w/24v - I agree it's marked that 24v is OK, but I also read instances where they have died when powered that high. Personally, I use a cheap buck converter and feed it 5v. The [...]

I have one 5V buck converter as well, tried it without the baseplate. Tested with multimeter, output is 5V, so no problem. Connected it to Vin, Ground & D4. Still same problem.

But i found out something crucial upon doing the above. Whenever i connect Vin & Ground Only (Without D4), i can access the said light in the WLAN app. But when i connect all 3, Vin, Ground & D4, i get "Device Unreachable".

So can i conclude that this is almost certainly a problem with the data line?

wchpikus, 2019-12-31:
Tm1903 or ucs1903?
This is differend chip..
For me, you can have problem with speed 800/400..

https://www.google.com/url?q=https://www.bestlightingbuy.com/pdf/UCS1903%2520datasheet.pdf&sa=U&ved=2ahUKEwjJ7fD3meDmAhVtmIsKHZo6ChAQFjAAegQICBAB&usg=AOvVaw0L5xRSVZ746zXwNXkb4jz9

noblesse1234, 2019-12-31:
> Tm1903

24V UCS1903, 1 controller per 6 LED.

Which problem do you refer to?

wchpikus, 2019-12-31:
Check speed of data...
This chip can work with 400 or 800kHz...pin SET..

noblesse1234, 2019-12-31:
Does that mean it will not work effectively when used with WLED? Sorry i am new to this. I don't seem to have a problem though when it first worked with WLED when changing effects.

What solution do i have to fix the data line?

wchpikus, 2019-12-31:
This chip can be used with two speed od data incomming.
What is selected in firmware i do not know, but in hardware (chip on strip) you can setup speed of data..400 or 800kHz.
This pin (set) is for that (look for datasheet).
If connection is ok, you have good software, you can do few things.
One- setup speed.
Two- cross this chip,power strip as 5V and check it.

noblesse1234, 2019-12-31:
> This chip can be used with two speed od data incomming.
> What is selected in firmware i do not know, but in hardware (chip on strip) you can setup speed of data..400 or 800kHz.
> This pin (set) is for [...]

Sorry how do i setup the speed of data? I don't understand Chinese actually. What does pin (set) mean? Would be grateful if you could point me to a video/link explaining how to do it.

Will try powering it in 5V in a moment.

Def3nder, 2020-01-08:
Hi @noblesse1234 if the SET pin is connected to VDD, it's working as "low-speed model" (400 Kbps), if the pin is not connected at all (floating) it's working as "high-speed model" (800Kbps).
[image]

You can use a multimeter and measure if there is a connection between SET and VDD or not 😄

noblesse1234, 2020-01-11:
> Hi @noblesse1234 if the SET pin is connected to VDD, it's working as "low-speed model" (400 Kbps), if the pin is not connected at all (floating) it's working as "high-speed model" (800Kbps).
> [image]
> 
> You can use a multimeter and measure if there is a connection between SET and VDD or not 😄

Apparently no, thanks for the guide.

I went ahead and bought some SN74AHCT125N, but i am clueless on how to use them.

I came across this site but it's Rasp Pi, not NodeMCU:

https://learn.adafruit.com/neopixels-on-raspberry-pi/raspberry-pi-wiring

1) Pi GPIO18 to 74AHCT125 pin 1A
2) 74AHCT125 pin 1Y to NeoPixel DIN
3) Power supply ground to 74AHCT125 ground
4) Power supply ground to 74AHCT125 pin 1OE
5) Power supply ground to Pi GND
6) Power supply ground to NeoPixel GND
7) Power supply 5V to 74AHCT125 VCC
8) Power supply 5V to NeoPixel 5V.

From the above, do i do according to these?

1) NodeMCU D4 to 74AHCT125 pin 1A
2) 74AHCT125 pin 1Y to my strip's Data In
3) Power supply ground to 74AHCT125 ground
4) Power supply ground to 74AHCT125 pin 1OE
5) Power supply ground to NodeMCU GND
6) Power supply ground to my strip's GND
8) Power supply 24V to my strip's 24V

Thanks.

sr1sws, 2020-01-11:
You need +5V to Vcc on the 74AHCT125. All grounds need to be tied together (I think you're doing this).  I didn't detect any other issue, but I'm far from expert or practiced on this stuff.

Personally, I used this level shifter board: https://smile.amazon.com/gp/product/B07LG646VS/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Good luck!

Def3nder, 2020-01-13:
@sr1sws the 74AHCT125 is only from 3,3V to 5V (not the other direction) but the function in this scope is the same.

Hi @noblesse1234 ,

if I read the UCS1903 datasheet right, you can use data voltages down to 0,4V for operation.
So, you should be able to directly attach the DIN pin to D4 of the NodeMCU.

Basicaly you need to add 24V from the Power supply directly to the strip (VCC + GND) and the D4 pin of NodeMCU to the DIN pin of the strip and (important) the GND-pin of the NodeMCU to the strip GND pin.

The power the NodeMCU like you want.

This should work.

noblesse1234, 2020-01-21:
> @sr1sws the 74AHCT125 is only from 3,3V to 5V (not the other direction) but the function in this scope is the same.
>
> Hi @noblesse1234 ,
>
> if I read the UCS1903 datasheet right, you can use data [...]

It worked on the shorter trip (30") but not on the 6ft one.

So i tried to use 74AHCT125 but to no avail as well. I am wondering if i connected it wrongly? Attached is the picture.

[image]

[image]

What i did was 

1) 24V PSU to Strip (Power & GND)
2) Strip (Power & GND) to 24V-5V converter
3) Converter to Vin & GND of NodeMCU
4) Converter 5V to 74AHCT125 VCC
5) Strip Data to 74AHCT125 pin 1Y
6) NodeMCU D4 to 74AHCT125 pin 1A
7) Power supply ground to NodeMCU GND
7) NodeMCU GND to 74AHCT125 ground
8) NodeMCU GND to 74AHCT125 pin 1OE

Thanks.

Def3nder, 2020-01-21:
Hi @noblesse1234 , if a short strip does work well than you do not need the 74AHCT125 as the first pixel does exactle the same (the output voltage of each pixel is 5V).

So if the short strip is okay and the long one not, then it should be something with the power down the strip. 

Hmmm - but you wrote that the "long" one is 6 ft ?
This is about 3 meters - this would work with 5V strips without a problem, so 24V should get problems only with far longer strips.

The "shorter" strip: has this been exactly the same strip-type ?
Or has anything else changed ?

noblesse1234, 2020-01-21:
> Hi @noblesse1234 , if a short strip does work well than you do not need the 74AHCT125 as the first pixel does exactle the same (the output voltage of each pixel is 5V).
>
> So if the short strip is [...]

I tried two different PSU, one is a generic plug PSU, another is a new Meanwell PSU.

Tried 5V and 12V as well (12V on different 6 baseboards - the baseboard accepts up to 24V and converts to 5V for the NodeMCU).

Shorter strip is of the same brand strip. 

Does data out and data in position matters? I am powering the strip from one end only and that end starts with DO.

Def3nder, 2020-01-21:
The direction where you put the power in does not matter.
What matters is the direction of the data.

NodeMCU D4 needs to be connected to DI - DO would not work at all.
So, essentially, the DO PIN at the end of the strip should float and only the DI PIN should be connected to the NodeMCU.

Did you do it this way ?

noblesse1234, 2020-01-21:
The data enters from NodeMCU to DO first. I have attached a picture. The green line is the data line.

[image]

Does that mean the NodeMCU data line has to be moved to the other end of the strip where DI begins first?

Def3nder, 2020-01-21:
no, it's correct the way you did it!

The signal has to "flow like the arrow displays it" - the first "DI" on your strip is below the white tube far left on the picture. Then you have every 6 LED pixels another DO and a "cutting-space" where you can trim the strip and another DI again .... repeating until the end of the strip.

So the connection is okay.

noblesse1234, 2020-01-21:
hmm...then what's the problem if i didn't do anything wrong, what other parts should i diagnose?

AalianKhan, 2020-04-20:
@noblesse1234 Hey I am having the same problem. have you found a fix?

noblesse1234, 2020-04-20:
> @noblesse1234 Hey I am having the same problem. have you found a fix?

unfortunately no, let me know if you found a solution :(

AalianKhan, 2020-04-20:
I ordered a level shifter. I am not 100% sir it will work at all but it's worth a try. I will update you if it works

closed 2020-08-27 by stale[bot]