wled/WLED #541: Feature request: Candle effect that is individual on the LED's

fribse, 2020-01-02:
WLED: 0.9.0-b1

[Candleholder with WS2812B LEDs](https://community.home-assistant.io/t/modifying-my-swedish-candleholder/159590?u=fribse)

The current 'candle' effect is quite nice, but it looks really odd that all the 'candles' flicker synchronius.
The effect would be much improved if they flickered individually.

fishbone-git, 2020-01-02:
The effcalled Fire Flicker

fishbone-git, 2020-01-02:
The effect you want is called Fire Flicker

Aircoookie, 2020-01-02:
That candleholder is a really cool project! Fire flicker would definitely work, but it's effect is simpler and not as nice looking as Candle. I will work on a multi candle mode soon! For now, you could also create segments (1 for each LED) and set each segment to Candle. Save in preset slot 16 to easily recall the effect :)

fribse, 2020-01-04:
@fishbone-git  i did look at the fireflicker, but it didn't look as nice :-)
@aircookie, thankyou for the nice words about my project, the wife is starting to accept it, especially when I told here about the candle effect. Hey, I will look into the 'segment' thing, I thought they were synchronius as well :smile:

fribse, 2020-01-04:
@Aircoookie Ok, I got the segments set up, as it can 'only' handle 10, I set them up with a few LED's in each.
I can save them all to slot 16, but I can't make it put the effect on all of them when it's 'candle' I can do some of the other effects.
If I then apply the slot 16, it will do 'candle' on segment 0, but the others will just show solid.
Then if I click slot 1 after the slot 16 (slot 1 has a candle effect set up for 1 segment), it will start doing candle on all of the segments, and they are now out of sync.
Did I do something incorrcetly, or is it a bug?

Aircoookie, 2020-04-22:
Added in db60309ce8fdd4a793548f36ab1d03ace49019ec :)

closed 2020-04-22 by Aircoookie

fribse, 2020-04-23:
Weheeee 😁 thank you @Aircoookie