wled/WLED #542: ESP32 & APA102

wardwygaerts, 2020-01-02:
Since there is a different binary for APA102 with the ESP8266, does an APA102 strip also work with ESP32?
Is a ESP32 performorming better?

Def3nder, 2020-01-03:
Hi @wardwygaerts ,
yes, APA102 is supported for the ESP32 as well.
Clock PIN is GPIO0 and Data PIN is GPIO2.

tonyn0, 2020-01-03:
It needs to be modified just like the version for the ESP8266 (uncomment the APA part and recompile). My ESP32 board does not bring out GPIO0, so a pin-change is required as well for me.

Aircoookie, 2020-01-04:
Seems like a should be releasing a binary for APA102 ESP32 as well. Good point on GPIO0 on (at least some) ESP32 boards, @tonyn0 . What would be a sensible choice for default pins? I believe something that supports the QuinLed Dig Uno would be a great choice :)

tonyn0, 2020-01-04:
I planned on 2, 4, 5, and 18 to keep them together on one side.

wardwygaerts, 2020-01-05:
Ok, so I can't use the esp32 bin as provided? Is there and advantage to use an esp32?

tonyn0, 2020-01-05:
The standard ESP32 BIN uses GPIO0 for the clock and my board doesn't bring that out. Yours may.

I have seen mention of the ESP32 being more fluid in dimming (higher PWM frequency?).

Def3nder, 2020-01-06:
yes, the ESP32 has a way higher PWM frequency, but:
a) this is only relevant for solid (analog) RGB stripes (where the whole stripe has the same color)
b) this will only result in a more "flicker-free" light.
    The dimming resolution is the same as the birghtness is internaly stored in 8 bit.

For the APA102 this has no influence - PWM is not used for APA102 as this stripe is controlled digitaly.

rlust, 2020-01-13:
I have a ESP32 WROOM32, in the standard bin what pin would I use for clk?

Def3nder, 2020-01-14:
Hi @rlust the standard PINs are GPIO2 for the Data and GPIO0 for the CLK.

You can change this in NpbWrapper.h in line 23ff:
```
#if defined(USE_APA102) || defined(USE_WS2801) || defined(USE_LPD8806)
 #define CLKPIN 0
 #define DATAPIN 2
```

rlust, 2020-01-15:
Works thanks! used pin 2 on ESP32, found out that the length
of the connection to first LES makes a difference. Tried to use a 5ft tenth of cable and it took out the ESP. Ordered a level shifter.

Def3nder, 2020-01-15:
Hi @rlust ,
you can also use one cut-off LED as a level shifter: connect one single LED pixel close to the ESP pin and then begin with the longer wire to the strip.
Each LED shifts the data signal to 5V,too.

rockman55, 2020-01-28:
How can I compile the WLED code to be used with APA102 led's and QuinLed-Dig_uno?

Def3nder, 2020-01-29:
Hi @rockman55,
first, choose your ESP and enable the right `default_envs` in the in the section `[platformio]`.
Then uncomment line 7 in `NpbWrapper.h`:
`#define USE_APA102  // Uncomment for using APA102 LEDs.`
and change line 24 from 
` #define CLKPIN 0`
to
` #define CLKPIN 1`
The `DATAPIN `is per default set to use `GPIO2` (see line 25), so you need to set the `L1_D jumper` on the QuinLed-Dig_uno board to `GPIO2` (see [here](https://quinled.info/2019/05/09/quinled-dig-uno-pinoutwiring-guide/)).
Connect your strip with `DATA (SDI) to L1_D` and `CLOCK (CKLI) to L1_C`.

That should be all 😄

rockman55, 2020-01-29:
Thanks for this info. I am not sure how to recompile this and change to a bin file. Can you give me instructions on this or if I can just download via Arduino IDE? Thanks again for your help.

Rockman55

Def3nder, 2020-01-29:
Hi @rockman55, getting everything to work with the Arduino IDE is way more complicated, because you need to have exactly the library versions stated in the platformio.ini to get the same results.

Try to install VSCode and then the PlatformIO extension - that does work without any configuration "out-of-the-box" 😄 

Up to then here are two binaries: one for an Wemos D1 Mini and on for the ESP32 version:
[WLED pre0.9.0b2 QuinLed-Dig_uno ESP8266.zip](https://github.com/Aircoookie/WLED/files/4129284/WLED.pre0.9.0b2.QuinLed-Dig_uno.ESP8266.zip)
[WLED pre0.9.0b2 QuinLed-Dig_uno ESP32.zip](https://github.com/Aircoookie/WLED/files/4129283/WLED.pre0.9.0b2.QuinLed-Dig_uno.ESP32.zip)

...but I personally do not know how to flash the ESP32 version "by hand" - PlatformIO does this for me 😄

rockman55, 2020-01-29:
Can I use the binary file you have included(WLED pre0.9.ob2 QuinLed-Dig uno ESP8266.zip, after unzipping it) to work with a Wemos Di Mini without any compiling? Also what jumpers need to be set on the QuinLED-DIG_Uno?

Thanks again

Rockman55

Def3nder, 2020-01-30:
yes, you can - just download the ESP8266-Flasher from [here](http://www.dietrich-kindermann.de/Downloads/ESP8266-Flasher-x32-Installer.zip) and connect the WeMos D1 mini to USB and flash. After that you get an access point named "WLED-...." where you can connect with your phone to and configure the WiFi settings of the device - DONE 😄

rockman55, 2020-01-30:
Which binary file will drive the APA102 led’s, and which pins are connected?

Thanks,

 

PS sorry about all the questions but I have tried many scenarios.

Def3nder, 2020-01-31:
Hi @rockman55, for ESP8266 this [binary](https://github.com/Aircoookie/WLED/files/4129284/WLED.pre0.9.0b2.QuinLed-Dig_uno.ESP8266.zip) (see https://github.com/Aircoookie/WLED/issues/542#issuecomment-579852857) is for `APA102` with PINs `GPIO1 for clock` and `GPIO2 for DATA`.

See https://github.com/Aircoookie/WLED/issues/542#issuecomment-579673999 how to prepare the QuinLed-Dig_uno for this.

rockman55, 2020-02-04:
Thanks for all your help. Got it working great, I used the bin file WLED pre0.9.0b2 QuinLed-Dig_uno ESP8266.bin.

Rockman55

closed 2020-06-10 by stale[bot]

rockman55, 2021-01-30:
I had previously posted comments regarding WLED binary files and APA102 led's. I have been using the provided bin file and was trying to update the file to a newer version of WLED and I cannot get a working file from the AirCookie to upgrade via OTA. Is there some other way to update this? I am using a Quin DigUno preassembled board. (See Below for previous post)

Hi @rockman55 <https://github.com/rockman55> , getting everything to work with the Arduino IDE is way more complicated, because you need to have exactly the library versions stated in the platformio.ini to get the same results.

Try to install VSCode and then the PlatformIO extension - that does work without any configuration "out-of-the-box" 😄

Up to then here are two binaries: one for an Wemos D1 Mini and on for the ESP32 version:
WLED pre0.9.0b2 QuinLed-Dig_uno ESP8266.zip <https://github.com/Aircoookie/WLED/files/4129284/WLED.pre0.9.0b2.QuinLed-Dig_uno.ESP8266.zip>
WLED pre0.9.0b2 QuinLed-Dig_uno ESP32.zip <https://github.com/Aircoookie/WLED/files/4129283/WLED.pre0.9.0b2.QuinLed-Dig_uno.ESP32.zip>

...but I personally do not know how to flash the ESP32 version "by hand" - PlatformIO does this for me 😄