wled/WLED #543: Switch on/off via homeassistant integration does not cross fade

c-soft, 2020-01-02:
I have my WLED lights installed in Home Assistant using official WLED integration.
I have 750ms crossfade setting enabled. 
When turning on/off via Home Assistant the crossfade does not happen - it turns on/off immediately. 
Using WLED UI works fine. Using button works too.

Am I doing something wrong, some configuration issue or is it a bug?
Thank you!

Aircoookie, 2020-01-04:
Hi!
I've been seeing that behavior as well, but I believe it is intended and/or settable in HomeAssistant, defaulting to no transition. @frenck definitely knows more than I about that :)

frenck, 2020-01-04:
The HA default behavior is not doing transition times, which the integration mimics. However, it Home Assistant does allow for specifying a transition time on a service call.

So for example, you can do a automation that triggers a change in your WLED light, and in that action, you can specify a one-time transition time, e.g., 10 seconds. Just for that single transition/action/automation call, the transition time will be 10 seconds.

This allows for greater flexibility. For example:
- On alarm, turn on all lights, full red, instantly.
- On movie time, player starts, dim LED lights slowly over a longer transition time.

c-soft, 2020-01-04:
Thank you for explanation, I see your point.

Having said that, I would still argue that maybe it makes sense to use the transition time set in WLED options when turning on/off from Homeassistant. Here's why:

- **consistency of behavior between switching the light on/off from HomeAssistant UI and physical button**. I my case, I use led strip as kitchen lamp (with button), and it fades nicely when turning on/off. That's not the case if i do it via button on HomeAssistant's UI. 
- **consistency with MQTT** - if you turn on WLED via MQTT, it currently uses the transition effect
- **other lights** - for example quite popular Fibaro dimmers  have similar setting (transition time), and it works no matter if the light is switched via button or remotely

If I may suggest an idea here, it would be nice if transition time set in settings would be used as default **unless** specified otherwise in service call.

What do you think, does it make sense?

Also, I did some testing and it seems like at the moment transition time parameter does not work for me - transition always takes about 1 second. Is it a genuine bug or are you seeing something else?
Thank you!

frenck, 2020-01-04:
Could be, please file an issue @ the Home Assistant repository with reproduction scenario's, since this is not the correct repository to report issues or suggestions for Home Assistant.

Thanks 👍

c-soft, 2020-01-04:
Thanks, I've entered two issues:

The transition time not being respected issue:
https://github.com/home-assistant/home-assistant/issues/30488

... and my suggestion to the change of the behavior:
https://github.com/home-assistant/home-assistant/issues/30489

I'm closing this one then as this is the wrong forum.

closed 2020-01-04 by c-soft