wled/WLED #544: Support for GE Color Effect

chuckj, 2020-01-02:
As a user of GE Color Effect strings, I am wondering if you've considered or had plans to support GECE protocol in WLED?

Aircoookie, 2020-01-04:
We could consider that as part of the rewrite to the `NpbWrapper.h`. Technologically, these are inferior to normal WS-type LEDs (max. 50 LEDs, only 24fps refresh, 12bit colors), but having them supported would definitely be nice!

udummy, 2020-12-14:
I know this post is half as old as Moses, But I have a lot of strings of GECE nodes.  They are slow, but I would love to see this option as well, no rush.  Blessings

friedpope, 2021-12-13:
Thought I'd throw in my upvote on this too!  I have three GECE strings running ESPixelStick firmware on ESP-01s, I'd love to get these natively on WLED.  Since my last look at the firmware I noticed the DDP output option in WLED, I'm planning to give that a try in combination with the ESPixelStick firmware and see how well I can get a master WLED station to sync up all of the GECE strands.