wled/WLED #545: Issue with same color in 2 tones.

rubenzori86, 2020-01-02:
Issue with same color in 2 tones. I have setup on White but i get this. 

[image]
[image]

Biothefirst, 2020-01-03:
I recommend that you don't use white dead center. I have had odd issues with it. 

Also is it a gradual change or drastic like the pic. Ie is the pic doing justice. 
If so you might be having voltage loss along your string. Should not though for how short it is. Or your signal line is got an issue where it changes color. 

Do you have a multi meter or a scope?

Also don't forget to toss a coin to your Witcher.

raoulteeuwen, 2020-01-03:
what is shown in the Segments-tab?

rubenzori86, 2020-01-03:
I configured wled on settings according with my power supply and now i think os working better

Aircoookie, 2020-01-04:
Seems like there were either (unintentionally) two segments as @raoulteeuwen suggested or there is a significant voltage drop at that point in your strip, which could happen with bad connections or solder joints. Glad it is working better now :) Feel free to reopen if the issue happens again.

By the way, I will finally publish a new binary in a few days that will get rid of the pesky success typeError :)

closed 2020-01-04 by Aircoookie