wled/WLED #546: Homenassistant integration disconnects

rubenzori86, 2020-01-02:
I have beta release and used default ha integration. But light disconnects recurrently every  x time
[image]

kabelux, 2020-01-03:
Same here. Esp32 is really near my router

rubenzori86, 2020-01-03:
Finally i decided setup with mqtt... Native integration doesnt work fine

Aircoookie, 2020-01-04:
Hi!
I believe this is rather a general connectivity issue than specific to the HA integration, since that just interfaces to the JSON API. @rubenzori86 are you using ESP32 as well? And did you compile yourself (if yes, which arduino core and LwiP) or did you use a binary (if yes, which)?

Thank you!

rubenzori86, 2020-01-04:
Hi, I'm usign Wemos mini D1 and I flashed directly compiled binary.

Dreamoffice, 2020-01-05:
same here. Wemos D1 mini with the compiled file. it disconnects oftem from home assistant and in the UI i got this message.....
[image]

raoulteeuwen, 2020-01-05:
1. @Dreamoffice how do you establish WLED/the Wemos is disconnecting often from HA?
2. And how did you integrate WLED into HA? Via HA > Configuration > Integrations > Add(+) > WLED > enter ip-of-WLED ?

Dreamoffice, 2020-01-05:
1. i can see it in the state history that it is not available. 
2. yes i did it exactly this way. 
few minutes ago i was playing around with the segments. it was not possible to set up segments. i got this errors in the UI:
[image]
[image]

Aircoookie, 2020-01-06:
@Dreamoffice ah yes, this issue happens when you run out of memory. Unfortunately, requests allocate double the required memory in 0.9.0, which means it happens quite often. Already fixed in source, new release this weekend!

Dreamoffice, 2020-01-06:
i swaped the system from D1 Mini to nodemcu, it is much more stable, but the police all effect is not working with 816 LEDs it looks more like a sparkle effect.

Aircoookie, 2020-01-06:
@Dreamoffice could it be that you have more than the default segment? ("Segment 0")
Then on segment would draw over another and interfere with most effects. If you'd like to have multiple Segments, make sure they don't overlap :)

mdietinger, 2020-01-16:
Had as well huge connectivity issues with WLED and Home Assistant. UI of native WLED app was very slowly loading. Resolved it by changing Wifi channel to 11.
This improved reponsiness dramatically.
You need to check with a WLAN analyser App what works best for you, as well ensure you are not interfering with Zigbee frequencies.

rubenzori86, 2020-05-15:
Same is working with last updates

closed 2020-05-15 by rubenzori86

c-soft, 2020-10-04:
I'm seeing the same issue with latest Hassio 0.115 and Wled 0.10.0  (I'm using  D1 mini) :-(