wled/WLED #547: New SK6812 RGBW Setup - LEDs appear dead, NodeMCU doesn't boot when D4 connected.

alancoleman1972, 2020-01-03:
Hi

I've purchased a 1M 144 LED RGBW SK6812, 5v8A PSU and ESP8266 ESP-12E NodeMCU.  Flashed the ESP8266 fine, connectivity from andorid app working.   PSU is powering both ESP8266 and LEDs.

The LEDs appear totally dead.  If I power the setup with D4 connected to the LEDs the ESP8226 is unavailable; the blue power led shows but I notice that it doesn't blink when power is applied which it does without D4 connected.  Applied power to both ends of LED strip, still the same.  Is there a way to identify if the LED strip is faulty?  

Greatful for any advice
Alan

raoulteeuwen, 2020-01-03:
Hi @alancoleman1972 . Afaik, a 8A power supply is a little small for 144 LEDs: 0.06 W/led (on full white) * 144 = 8.64A. But i would assume you should get light anyway. Could you share how you connected all parts? Also what bin file exactly you flashed to your NodeMCU? Just wondering whether D4 is the right pin for you? Is it easy to just try different data pins and see what that brings?

alancoleman1972, 2020-01-03:
Hi @raoulteeuwen, thanks for getting back to me.  I recall the app advising a 9A PSU but I don't expect to ever have the LEDs anywhere near full white so thought 8A would do (if i can ever get them lit!).

I've just flashed WLED_0.9.0-b1_ESP8266.bin and it's still behaving the same.  I tried D0-D8, ESP8266 powers up when D0,D1,D2,D5,D6,D7,D8 but no blue light with D3 and although it seems to power with D4 connected I cannot access the device.

I've attached a piccy of my (test) setup.  The yellow wire is the data connected the the middle wire on the SK6812 as I can't find my green one.
[image]

Thanks
Alan

raoulteeuwen, 2020-01-03:
So you have a working responsive NodeMCU with WLED as long as your strip is not also connected, but as soon as you connect the strip the NodeMCU becomes unresponsive? So https://bbs.espressif.com/viewtopic.php?t=4597#p15844 should be the problem?

Could you connect the ground of both the strip as well as NodeMCU on the same pin of the power supply? I assume that doesn't make a difference, but please check.

Just to make sure: you are feeding the data from the right side (there should be an arrow on the strip indicating flow direction of data)?

raoulteeuwen, 2020-01-03:
(didn't know what the normal behaviour of the blue LED is, just read at https://www.esp8266.com/viewtopic.php?p=64837: "The Blue LED on the ESP-12 module (module used in NodeMCU) is connected to the Serial TX of the ESP8266 in reverse bias, so if Serial TX goes Low then the Blue LED is on. It blinks when you power on the board because the ESP8266 spits out it's boot loading message at 74880 baud on startup causing highs and lows. The Serial TX just so happens to be the same pin as D4, so if you are bringing that Low then it will turn on. See the Pinout diagram for the NodeMcu t...")

alancoleman1972, 2020-01-03:
Wow, thanks.  Finally geting somewhere now.  I wasn't quite sure what the arrows on the strip meant, i thought it weird that arrows where on the same side as the ground label, so having now changed the D4 from the NodeMCU to the other end of the strip I at least get some lights.

So now the next issue.  The lights don't respond correctly to the config.  When it powered up I got maybe 5 or 6 LEDS light up, sort of random locations but all in first 69 LED range.  I can control the colour and brightness of these and if I choose an FX mode a few other LED light up.

I tried downgrading to v0.8.6 and when it restarted the same LEDS light up but then the last 72 lit up red.  I power cycled and then it returned to the same as with v0.9.0-b1 but the last 72 didn't light.

I then downgraded to v0.8.5.  Same result until i moved the FX2nd Color slider at which point the non-lit LEDS below 69 all light up in a different colour.  The remaining 75 stay off.

I've attached a piccy but its quite difficult to see the few that are different colours.

Thanks for helping
[image]

raoulteeuwen, 2020-01-03:
Did you see the option to specify the nr of LEDs in Settings? This tech is not able to auto-detect how many LEDs are attached.

alancoleman1972, 2020-01-04:
Yeah, I've had the number set at 144 the whole time.  Interestingly with WLED_0.9.0-b1_ESP8266.bin if I set the segment to, for instance, 0-8 then LED 7 still lights and i can change the colour but the other few that were lit and controllable stay lit but the colours dont change.  I'm not sure what to try next other than buy a new NodeMCU.

raoulteeuwen, 2020-01-04:
I read you've found the Segments feature; even when you delete all segments, still not all LEDs light up?

alancoleman1972, 2020-01-06:
I don't appear to be able to delete all segments.  There appears to be a default which I can't remove, should it be possible to have 0 or must there always be min of 1?

Aircoookie, 2020-01-06:
No, one segment will always remain :) If you could delete it, all LEDs would turn off and nothing but adding new segments would work

alancoleman1972, 2020-01-10:
Still no progress.  I will try and get the LEDs attached to a spare Raspberry PI and see if I can get them working with the CIrcuitPython libraries.

Def3nder, 2020-01-14:
Hi @alancoleman1972 , do you have the possibility to cut-off one pixel from the LED strip ?

...and then connect just this one pixel ? If the NodeMCU does not boot when only connecting something to it it's most likely a matter of protection because of "too much current".

alancoleman1972, 2020-01-14:
Hi @Def3nder,  I have resolved the issue of the NodeMCU not starting - it was when I was connecting the LED data strip data the wrong way round!  The problem I'm trying to resolve at the moment is that only a few of the 144 LEDs are controllable (maybe 5 or 6 randomly distributed along the strip).

Def3nder, 2020-01-14:
Hi @alancoleman1972 , did you check the "use RGBW" flag (now named **"LEDs are 4-channel type (RGBW)"**) in the LED settings ?

...if the strip expects 4 bytes per LED (RGBW) and receives only three (RGB) this looks kind of "random RGB colors".

alancoleman1972, 2020-01-14:
Hi @Def3nder, Yeah, I made sure the LEDS type was set.  I also tried unselecting and got even stranger results!  I'm struggling to make time to temporarily repurpose a raspberry pi to see if the LEDs work on a different platform so I've ordered a new NodeNCU to try in the meantime.  Hopefully it arrives before weekend so I can test again.

alancoleman1972, 2020-01-16:
Well, the LED strip is now fully working.  My new NodeMCU arrived so I flashed it with WLED_0.9.0-b1_ESP8266.bin and instantly worked when connected to the strip.  I guess the original device had a fault somewhere.  Thanks to everybody for your assistance.  Very much appreciated.

Def3nder, 2020-01-20:
@alancoleman1972 ,

if the issue is solved, please close it - so the counter of "open issues" decreases again 😄

closed 2020-01-20 by alancoleman1972