wled/WLED #548: Alexa turns WLED off but not back on (but reports it is on)

mcpacific, 2020-01-03:
I flashed NodeMCU with WLED version 0.9.0-b1. It was working fine until I added a macro to 'Slot 2'. Removing the macro had no effect. Not sure if that is why it stopped working properly.

Alexa reports the light being 'On' but it is still off. I can turn it off and on with teh WLED web interface; if I turn it on this way, I can turn it off with Alexa... it just won't turn back on.

Aircoookie, 2020-01-04:
Hi!
Could you check your time and macro settings? Maybe something is set for "Alexa On macro".
If that doesn't help, you could try deleting the WLED light and rediscovering in the Alexa app :)

dkneisz, 2020-01-11:
I ran into an issue when I tried to turn on my WLED but no light went on. The problem was cused by the fact, that I turned of the lights using the "Timer", which set the brightness to 0. After turning on the light it was on, but with 0 brightness. Did your macro by any chance set the brightness to 0?

mcpacific, 2020-01-16:
> Hi!
> Could you check your time and macro settings? Maybe something is set for "Alexa On macro".
> If that doesn't help, you could try deleting the WLED light and rediscovering in the Alexa app :)
> Also, what Echo Device are you using?

Alexa On Macro was set for #1 of defined macros, which is defined as 'FX=15&SX=100&R=0&G=139&B=0&R2=0&G2=255&B2=0' (without the quotes). I deleted the WLED light inside Alexa twice and re-discovered but that doesn't seem to help.

However, after changing the Alexa On Macro to 0 it now works as expected. Does this mean we cannot define specific macros for Alexa?

Aircoookie, 2020-01-16:
@mcpacific Hi! The macro is in fact executed, but it replaces the default "Turn on" command and your macro doesn't contain the command for turning the light on. Try adding `&T=1` to your macro :)

mcpacific, 2020-01-16:
Great, all is working now, thank you! Also, I experimented with the master brightness as dkneisz commented and it also worked with that in the macro (without `&T=1`). Either way, it's all good! :)

closed 2020-01-17 by Aircoookie