wled/WLED #549: WS2815 power consumption?

stringandstickytape, 2020-01-04:
I'm using a 12V 6A power supply.  It powers a buck converter -> 5V -> ESP8266 NodeMCU, and a WS2815 strip: 5m x 60 = 300 leds.  An ammeter measures the total load (including the converter and NodeMCU).

WLED is set to maximum current 5A, LED type set to 12V (30ma).  

When the above is set to solid red, an inline ammeter shows 4.04A drain; WLED says Current estimated usage: 3400mA.

Set to solid yellow, solid magenta or solid cyan, the ammeter shows 3.25A drain; WLED says Current estimated usage: 4994mA.  

Set to solid white, the ammeter shows 2.5A drain.  WLED says Current estimated usage: 4990mA.

If I turn off the WLED brightness limiter entirely, then the ammeter says the drain is 4.04A no matter what colour or white I select.

This Doesn't Seem Right.

On extensive investigation, I found this likely explanation:

http://www.thesmarthomehookup.com/the-complete-guide-to-selecting-individually-addressable-led-strips/

> You can see the really odd behavior of the WS2815 when it comes to current draw.  Basically, a single pixel draws the exact same amount of current at 50% red as it does at 50% white, even though the white consists of red, green, and blue.  An awesome commenter in another one of my videos explained why, but the cliffsnotes is that each channel is powered in series instead of parallel, and if only a single channel is wanted the other two are shorted out by a transistor resistor combo to keep the current constant.

It would appear that a slightly different algorithm is therefore needed for the brightness limiter when using WS2815?  In my situation I can disable it, but only because I know my 6A power supply exceeds the maximum 4.04A that this setup will draw under any circumstance.

Happy to run more tests and report back if helpful, and thank you for your work on WLED which is a fantastic app and getting better by the day.

Aircoookie, 2020-01-04:
Hi,

thank you for your conclusive testing :)
This is indeed a very interesting design decision... And you are correct that it means that the WLED brightness limiter is totally inadequate for WS2815.

I should add a new calculation method based on the brightest channel of each LED for this type!

Aircoookie, 2020-01-18:
Closing since I've merged your PR for this :)

closed 2020-01-18 by Aircoookie