wled/WLED #551: Travis maximum log size reached

Aircoookie, 2020-01-04:
#550 failed because of the log size becoming too large with the extra platform.

Does anyone know how to reduce the log output level of platformio to potentially allow for more platforms?
Thank you!

fishbone-git, 2020-01-04:
adding -s to the build flags should do it, at least according to the docs.  Seems to have the same output for local builds so can't be certain.

Aircoookie, 2020-01-05:
Thank you :) Unfortunately doesn't seem to make a difference: https://github.com/Aircoookie/WLED/commit/e2caeeab9a8f8ba5a6512e7e974a076787ef6106
Where did you find that doc?

fishbone-git, 2020-01-05:
https://docs.platformio.org/en/latest/userguide/cmd_run.html
vscode calls platformio run to start the build.

fishbone-git, 2020-01-05:
perhaps remove the -v in your .travis.yaml file for the platformio ci command line:
script:
    - platformio ci --project-conf=./platformio.ini -v
This would produce verbose output.  Another option would be to redirect the install to /dev/null:
pip install -U platformio > /dev/null

The latter is a suggestion here:
https://travis-ci.community/t/large-pip-installs-exceed-maximum-log-length/1599

The former is documented here:
https://docs.platformio.org/en/latest/userguide/cmd_ci.html

Aircoookie, 2020-01-06:
Thanks a bunch! I was unaware of the verbose option set in travis.yml since I was not the one who initially set up the CI. And I was already wondering why the travis log is much longer than my local PIO environment. That did the trick!

closed 2020-01-06 by Aircoookie