wled/WLED #552: Support WWA channel type for SK6812-WWA type strips

billchurch, 2020-01-04:
Currently there is no support for WWA (white, white, amber) style strips. I plan on adding this and doing a PR, waiting for strips to test with first. This issue is just a placeholder.

Example strip: https://www.aliexpress.com/item/32430149013.html

Nobeernogman, 2020-01-04:
I will buy 1m SK6812WWA 60 IP67 from Ali :-) Will report back when it's arrived.

Aircoookie, 2020-01-05:
Hi! Those are in fact supported hardware wise (since they use the same protocol as RGB ones).
Red channel corresponds to amber, and green and blue to cold and warm white, I believe.

What I yet have to do is add explicit WWA support to the UI (as in a colorpicker that makes intuitive sense for white channels)

billchurch, 2020-01-05:
That was my thought, what would be needed is mostly UI. I don’t have a strip to play with but will grab one.

billchurch, 2020-01-05:
I feel like it might make sense to do sliders for each channel, and then maybe some preset color temps ?

Nobeernogman, 2020-01-06:
Just bough a SK6812WWA. Will report back when the mailman knocks at my door.

vmazmaz, 2020-01-08:
> I feel like it might make sense to do sliders for each channel, and then maybe some preset color temps ?

I feel a single slider for color temperature would be my input of predilection for WWA.

nicoduck, 2020-03-04:
> I feel a single slider for color temperature would be my input of predilection for WWA.

I would welcome an implementation similar to most other CCT lights. An option to optionally enable intensity correction would be nice too, otherwise setting it to a medium colour temperaure would enable both white LEDs at 100% which would make the light brighter than on cold or warm white.
The amber LED is not as bright as the white ones and would need its own slider. That's not a big surprise since most amber lights I know are mainly usable for ambient lighting.

Here is a short video of the strip presenting a simple pattern: https://youtu.be/hz0-mlSzlsU

kulps, 2020-07-14:
I would definitely appreciate WWA support in the UI. 
I'm going to predominantly using WLED to manage WWA strips. It's a minor convenience to see the colour in the UI as it's going to come out instead of basically just dragging it around until I find what I'm looking for.

m-s-e, 2020-11-10:
Following with interest!
WWA integration would be great.

MFlasskamp, 2020-11-10:
I'll second that!

james-fry, 2020-12-08:
This would be great! Just in the process of adding a 3 meter strip (288 SK6812s) in my office.
The strips work perfectly, just need the UI to show the correct colors in the color picker etc.

mikkihugo, 2021-05-16:
Would be really awesome to get as I have a couple of them

adub08, 2021-08-03:
I'd like to second this too, it would be great! :)

moscher, 2021-08-29:
that  would be a awesome feature!!

blazoncek, 2021-08-29:
As a workaround you can use RGB PWM setup and then change UI parameters to hide color wheel and use RGB sliders.
You can control each channel of your WWA using RGB sliders then.

zwevert, 2021-09-07:
I would highly appreciate support for sk6812 was as well. Thanks !

mattkerrison, 2021-12-20:
Any updates on this? I have a few strips of SK6812 WWAs and would be really nice to have a custom color wheel or sliders for the 3 different whites

blazoncek, 2021-12-20:
You can enable RGB sliders and disble color wheel in UI settings page and you will be able to control each channel individually until this is added to WLED.

fssup, 2022-05-09:
Would also love support for this.

mortelil, 2022-05-09:
Support for these would be great!

james-fry, 2022-09-29:
Any indication when (or if) this will be added?

blazoncek, 2022-09-29:
Not in any immediate plan.

maverick1872, 2022-10-22:
Is there a blocker for having this implemented or has no one undertaken opening a PR to introduce it?

blazoncek, 2022-10-22:
The latter.
Though it may need a *serious* talks with Home Assistant developers to recognise WLED device as such and accept it into their *integration*.

DjordjeMandic, 2023-09-28:
@blazoncek You mean that HA integration has to support color temperature (this ui feature were talking about) too? So no plans on working on this without implementing it in HA integration at same time?

blazoncek, 2023-09-28:
@DjordjeMandic  What happens in HA is HA's affair. WLED is not affiliated with HA and HA refuses to add CCT capable WLED device into integration (as of this writing).
WWA support is not implemented in WLED UI (or in core) and since none of the developers own WWA strip or have use for it there are no immediate plans to add such support.
But, as always, it is possible for someone with knowledge and willingness to implement such functionality and make a PR.

PR #4484 (2025-02-13): WWA strip support & parallel I2S for S2/S3 (bumping outputs from 5/4 to 12)
Bus wrapper modifications
- NeoPixelBus update 2.8.3
- automatic selection of appropriate single I2S bus (`X1xxxxxxMethod`)
- removed I2S0 on ESP32 (used by AudioReactive)
- renumbered internal bus numbers (iType)
- added buffer size reporting (prototype, requested @makuna to add appropriate methods to NPB)

Bus modifications
- WWA strip support
- bus initialisation rewrite (moved to `WS2812FX::finalizeInit()`)
- optional parallel I2S (ESP32, S2 & S3)
- ABL speed improvements (no float)

This modification will allow un [... cut]