wled/WLED #553: "Please select a valid value" in favorites preset time and transition fields

raoulteeuwen, 2020-01-04:
I'm using 0.90b1 on 8266-NodeMCU's. I've noticed when i visit the IP-address of the WLED NodeMCU and go to the favorites screen, in the fields for preset time and transition i see values like 19.7 and 0.7, they have a red box around them, and when i click in the field, i'm shown "Please select a valid value. The two nearest values are " and then i'm offered 2 values with a decimal. Rather odd?

I also have trouble changing what i have entered in that dialog: when i change the values to ommit the error, how do i save them? When i untick or tick (depending on what is shown when i start) the 'preset cycle' box, the values for preset and transition jump back to the values that cause a red box and the error. No 'save' button to press...

brentbrooks70, 2020-01-04:
@raoulteeuwen If the **Preset Cycle** is _ticked_, any changes made will be reverted to the previous state when it is _unticked_. Please _untick_, make changes and _re-tick_ to have them persist. I had similar issues and this seemed to fix it. As for the odd entries (19.7 & 0.7) in the time fields, mine kept giving me 12 & 1.2 but neither was ever red nor did I have any issue with using numbers with decimal places in either field. I am currently running 30-second cycles with 1.5-second transitions on 3xD1 minis, 2xLOLIN NodeMCU v3, 1xHuzzah Feather and 1xFeather 32 all running 0.90b1 commit bd4d1cd from 12/17/19. 

> No 'save' button to press...

This is handled via an onchange in line 732 of index.htm
`      <input type="checkbox" id="cyToggle" onchange="toggleCY()">`
so the changes should be "saved" as soon as the change occurs. 

I hope at least the unticking helps you going forward and good luck.

raoulteeuwen, 2020-01-04:
Thank you @brentbrooks70 , that works!

Aircoookie, 2020-01-05:
Thank you for the feedback!
I will change the toggleCY() to also send changes to the text-fields when unticked, as that causing the values resetting is not intuitive and was not my intention.

About the invalid number issue, that one is really strange. 19.7 and 0.7 are definitely legal values for this fields. Perhaps this is a locale issue i need to work out as some countries use `,` as decimal separator... What browser are you using and is it using a different language than English? 
I did just try this with a German Chrome installation and it displays commas as separators by default, but works as intented, letting me use either a comma or dot as decimal separator.
Also I'm curious as to what the two closest valid values the browser suggested were :)

Thank you for all your help!

raoulteeuwen, 2020-01-05:
I'm using Firefox. I get this both on a Samsung (s10) as well as on my laptop (Kubuntu). Firefox is using the English language. If no-one else is reporting it, don't put too much time in it.
For the 19.7 value i get the suggestion to use 19.2 and 20.2 :-). For the 0.7 i'm suggested 0 or 1. I'm given . as decimal separator.

Thank YOU for all the time you spend on this!!!

Aircoookie, 2020-01-05:
Ok, I tried out Firefox and getting this as well! Seems as Firefox handles a missing step attribute differently than chrome. Will fix in the next update I do to the UI!

closed 2021-01-08 by Aircoookie