wled/WLED #554: Alexa fails to re-recognize after update to WLED_0.9.0-b1_ESP8266_1M_ota.bin

bphick, 2020-01-04:
I loved the new effects I ran on my test NodeMCU last night, and decided to  upgrade my "production" boards with WLED_0.9.0-b1_ESP8266_1M_ota.bin this AM.  I noted the inability to have it save a boot state properly as documented elsewhere and apparently resolved.  However, Alexa would no longer see any of my upgraded devices.   I did look at the old info on this but it has been over a year since it was documented.  I do have the same names used before under the Sync Interface, Alexa Voice Assistant, Emulate Alexa divice (checked) and same invocation name.  Yet for the devices, Alexa will not recognize them anymore. Disappointed, I deleted the device and tried a rediscovery. No result, then rebooted all of my Alexa devices, router, reloaded the bin, etc.  Still no discovery anymore.  New boards discover immediately... but I'm out of new boards.  ;-)

brentbrooks70, 2020-01-04:
As of 0.8.6 the 1MB versions either support OTA **or** Alexa | Blynk | Hue.
I think you want the _WLED_0.8.6_ESP01_1m_full.bin_ to support Alexa but of course, you will not have OTA abilities. From the looks of it, OTA will not be supported on 1M boards going forward so either 

- do not update beyond 0.8.5 to keep both abilities
- use the Alexa features without updates
- update to 4M boards

Taken from https://github.com/Aircoookie/WLED/releases/tag/v0.8.6:
> Note regarding ESP01:
> ESP01 with 512k flash no longer supports Alexa, Blynk or Hue sync interfaces. If you use them, please do not update.
> **_ ESP01 with 1MB flash either supports OTA updates OR the above interfaces, just choose the respective binary._**
> From the next version onward, it will very likely be impossible for me to maintain 512k ESP01 support.
> ESP01 1M will stay supported, although OTA updates will not be possible.
> To get the most out of WLED, I highly recommend you to upgrade to a board with 4MB flash.
> Sorry for the inconvenience!

Hope this helps and good luck!!

bphick, 2020-01-05:
Thanks brentbrooks70: I had tried the full.bin file first, and couldn't seem to get it to work at all so I just moved to the next bin, once it worked, I configured it.  I'm not sure, but looking at the past documentation here https://github.com/Aircoookie/WLED/issues/3  I believe my issue to be similar, to that discussed here, but since that was a long, long time ago, I was looking more for confirmation that it's still a valid fix or if there is another way.  All newly installed boards work fine, even on the version mentioned above.  It's only (or seems to be) if there is a change to the initial install that gets hung up and will no longer recognize a previously used board.  None of my boards are ESP01s.  All are the level of NodeMCU Lion V3.

Hope this clarifies the issue.

brentbrooks70, 2020-01-05:
@bphick Sorry, I just assumed that if you were using 1M bins that you had a 1M board. If you have 4M boards then you should try WLED_0.9.0-b1_ESP8266.bin, this has everything in it. Not sure why this would help but you could try to completely erase the board and start over. If you are using Arduino to compile and upload under **Tools** -> **Erase Flash** select "**All Flash Contents**", or, you could try from WLED itself going to **Config** (used to be settings) -> **Security & Updates** check **Factory reset:**  and click **Save & Reboot**. You will then have to configure this from scratch including connecting to its AP and connecting WiFi, LED settings, etc..  If this doesn't help maybe @Aircoookie can weigh-in with some of his genius-level advice, I did see he was very active this morning.....

bphick, 2020-01-05:
I'll give that a try.  I was under the impression that the other two bin commits had MORE than the regular one.  I have them running effects now for the next few hours that people are seeming to enjoy, so I'll have to try the WLED_0.9.0-b1_ESP8266.bin tomorrow.  Thanks for the head's up.  At times it's hard to know what is in / the advantage of each file.  ... but they have worked so well, that I have not felt the need to do any compiling myself...  which is a testament to some good coding!! :-)

Thanks again.

bphick, 2020-01-05:
brentbrooks70 ... Well the suspense was killing me.  So I took one down and tried the other Bin that you suggested (like the other one I used with the previous version).  YOU WERE ABSOLUTELY CORRECT.  Alexa DID indeed find the device.   THANK YOU!!!  This issue is CLOSED!

closed 2020-01-05 by bphick