wled/WLED #555: Home Assistant RGBW config

CaffeinatedFunctionality, 2020-01-04:
Hopefully this is an easy one. I can't get autodiscovery to work for WLED for Home Assistant. I have a normal config setting in configuration.yaml, but it won't seem to work. Config is below. I am sure I'm doing something wrong. Anyways these are sk6812s and they work perfectly on the WLED app. I just want to add them to my assistant.

`light:
  - platform: mqtt
    name: 'Above Cabinet Lights'
    command_topic: '/wled/abovecabinet'
    payload_on: 1
    payload_off: 0
    rgb_command_topic: '/wled/abovecabinet/col'
    rgb_command_template: "{{ '#%02x%02x%02x' | format(green, red, blue)}}"
    color_temp_command_topic: '/wled/abovecabinet'
    brightness_command_topic:  '/wled/abovecabinet'
    white_value_command_topic: '/wled/abovecabinet'
    effect_command_topic : "wled/abovecabinet/api"
    effect_list:
    - "FX=0"
    - "FX=1"
    - "FX=2"
    - "FX=3"
    - "FX=4"
    - "FX=5"
    - "FX=6"
    - "FX=7"
    - "FX=8"
    - "FX=9"
    - "FX=10"
    - "FX=11"
    - "FX=12"
    - "FX=13"
    - "FX=14"
    - "FX=15"
    - "FX=16"
    - "FX=17"
    - "FX=18"
    - "FX=19"
    - "FX=20"
    - "FX=21"
    - "FX=22"
    - "FX=23"
    - "FX=24"
    - "FX=25"
    - "FX=26"
    - "FX=27"
    - "FX=28"
    - "FX=29"
    - "FX=30"
    - "FX=31"
    - "FX=32"
    - "FX=33"
    - "FX=34"
    - "FX=35"
    - "FX=36"
    - "FX=37"
    - "FX=38"
    - "FX=39"
    - "FX=40"`

raoulteeuwen, 2020-01-04:
Afaik (i've got 2 NodeMCU's with WLED in HA) you don't need MQTT. See https://www.home-assistant.io/integrations/wled/ . Just go to HA > configuration > integration > add (+) > WLED > type ip of your WLED device.

CaffeinatedFunctionality, 2020-01-04:
Thanks. That's perfect!

closed 2020-01-04 by CaffeinatedFunctionality

CaffeinatedFunctionality, 2020-01-04:
Yep set them up in less than 2 minutes. I'm very happy that exists. That's pretty awesome!