wled/WLED #556: APA104 led lights

rspaargaren, 2020-01-04:
I am trying to use the library with APA104 lights. These are a clone of the WS2812 leds and for fastled there is a library. As far as I can see in fastled they are coded in the same way as WS2811 as they have a slight different timing.
In the code I have tried to change the:
#define PIXELMETHOD NeoEsp8266Uart1Ws2813Method
But if I change it to Ws2811 I get an error compiling?
Any Idea how to use these leds or how to compile in platformIO

brentbrooks70, 2020-01-05:
@rspaargaren Do you happen to know if it uses identical protocol/transport as one of the supported chips from the underlying library [NeoPixelBus by Makuna](https://github.com/Makuna/NeoPixelBus)

Here are the supported chipsets, taken from [https://github.com/Makuna/NeoPixelBus/wiki](https://github.com/Makuna/NeoPixelBus/wiki):
- WS2811, WS2812, WS2812b, WS2813, APA 106, SK6812, & LC8812 (NeoPixels)
- RGB and RGBW (in various color order)
- APA102 (DotStars)
- LPD8806
- All one and two wire RGB pixels that use the same transport as any of the above.

If not, then you would need to request that he add it to the library before WLED could use them. 
Sorry I don't have better info

Aircoookie, 2020-01-05:
Hi! Did you try hooking them up with the default WS2812 code? It is compatible with WS2811, so theoretically APA104's should work without any code modifications!

rspaargaren, 2020-01-05:
Hi I did a bit of playing around with the various libraries for fastled and neopixelbus. I had the strip working for a couple of year with Hyperion and a 5v Arduino. I knew at the start I also had some problems. I found out that with a level shifter and an external power supply to the shifter it works... started with usb power but that is to weak. There is still a bit of jittering when changing the lights but I suppose this was due to temporary setup with long DuPont cables and bad connections. So this issue can be closed for now. Thanks for the fast response.

Aircoookie, 2020-01-05:
Awesome! Hope you get it working perfectly soon! Feel free to reopen if it turns out that APA104 needs special treatment as compared to WS2812 (which also have the flicker etc. without levelshifter for some setups)

closed 2020-01-05 by Aircoookie